AdnanHodzic/displaylink-debian #585: debian buster Kernel: 5.10.0-7-amd64 -- external monitors not found

mfwvo9, 2021-06-18:
Hi

First of all: thanks for the amazing work.

I believe I'm quite close to a running solution, nonetheless my external monitors are not detected. I have experimented with different settings in 20-displaylink.conf which you see commented, so far without success.

What stuns me is the output of `xrandr --listproviders` which gives 5 Sink but no Source Provider.

Any attempt to help greatly appreciated.

`-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.10.0-7-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.0+dfsg
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [103c:8723]
VGA: Intel Corporation Device 9b41 (rev 02)
VGA (3D): 
X11 version: 1.20.11-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
 	Identifier "DisplayLink"
# 	MatchDriver "evdi"
 	Driver "modesetting"
	Option "PageFlip" "false"
# 	Option "AccelMethod" "none"
EndSection
# 
# Section "Device"
#     Identifier  "Intel"
#     Driver      "intel"
# EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xe0 cap: 0x2, Sink Output crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0xb8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x96 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x74 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

Barabazs, 2021-06-19:
You have to associate the providers, as noted in the [post-install-guide#display-detection](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection).

mfwvo9, 2021-06-19:
Yeah, sorry, I was not clear enough. That's what I meant with the missing Source output. You have that in the post install guide where my listproviders call lists this:

```
#> xrandr --listproviders 
Providers: number : 5
Provider 0: id: 0x47 cap: 0x2, Sink Output crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0x4ac cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x48a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x468 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x446 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```
So it's 'Sink Output' only. I don't know if this is important, it just caught my eye.

When I try to set the output source I get:
```
#> xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x47
  Serial number of failed request:  19
  Current serial number in output stream:  20
```

Barabazs, 2021-06-19:
I totally missed that part. Sorry about that.

Could you try with this config please.
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection`
```

mfwvo9, 2021-06-19:
That's actually the automatically generated config. Now it looks like this again:

```
$ cat /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

```

and the providers are gone:
```
± sudo ./displaylink-debian.sh --debug
[sudo] password for aherceg: 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.10.0-7-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.0+dfsg
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [103c:8723]
VGA: Intel Corporation Device 9b41 (rev 02)
VGA (3D): 
X11 version: 1.20.11-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

Now the providers are gone again.

Barabazs, 2021-06-26:
Any luck on your set-up?

mfwvo9, 2021-06-30:
No the result from my last comment remains unfortunately... I'm out of ideas.

mfwvo9, 2021-07-06:
Got it working!!!

1. In BIOS I had to activate "Enable High Resolution mode when connected to a USB-C DP alt mode dock" option.
2. In 20-displaylink.conf I had to replace Section "Device" by Section "OutputClass"

The debug output now looks like this

```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.10.0-7-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.0+dfsg
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [103c:8723]
VGA: Intel Corporation Device 9b41 (rev 02)
VGA (3D): 
X11 version: 1.20.11-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
	Identifier	"Intel"
	Driver		"intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 4 name:Intel
Provider 1: id: 0x124 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x102 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xad cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```
and life is great again.

Thanks a lot for your help!

closed 2021-07-06 by mfwvo9

Eddie-064, 2021-07-12:
HI @mfwvo9 I have some issue on debian buster #591.
where is your debian image download from?

mfwvo9, 2021-07-13:
Hi @Eddie-064 

I'm using official debian mirrors but it's not buster exclusively but a lot of packages from bullseye as well, including the kernel

Eddie-064, 2021-07-14:
thanks @mfwvo9 
I use official debian too, but the kernel version is only 4.19.
Could you provide the download image?   
and the debian  image I use is debian-10.10.0-amd64-DVD-1.iso

mfwvo9, 2021-07-23:
I don't know which Version of the image I used for installing. Sorry