AdnanHodzic/displaylink-debian #587: Displaylink port to Esxi possible ?

Docop2, 2021-06-19:
Hi  So as displaylink get quite issues with virtualization before, with win10 all was resolve and if you passthrough the d6000 or other it pop up perfectly. In ubuntu it's not working and with your project here, while it do install perfectly, don't give error, the eid and lsusb show up all port of the unit. Into:  xrandr --listproviders it don't show up, only the vmware-svga driver is actually present and no other show up in the xrandr gui.
 So was it something that you have tried, or it's only an issues with the core of displaylink not being capable... or a work around could do something ?
   On Esxi 6.0.    Thanks in advance

Barabazs, 2021-06-26:
Hi @Docop2. Your request is beyond the scope of displaylink-debian.

closed 2021-06-26 by Barabazs

Docop2, 2021-06-26:
actually it's fully for displaylink-debian.  But if the debian is in a virtualbox or vmware .. it don't pass.