AdnanHodzic/displaylink-debian #588: Setting up Environment variable SYSTEMINITDAEMON

ramifications123, 2021-06-28:
It's asking me to set up the environment variable in the title. If you help me, I'll donate some coffee money! Cheers.

Error
====================================
Installing driver version: 5.4

DisplayLink Linux Software 5.4.0-55.153 install script called: install
ERROR: the installer script is unable to find out how to start DisplayLinkManager service automatically on your system.
**Please set an environment variable SYSTEMINITDAEMON to 'upstart', 'systemd' or 'runit' before running the installation script to force one of the options.**
Installation terminated.
=====================================

Debug
=====================================
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-13-amd64

---------------- DisplayLink info ----------------

Driver version:
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
DisplayLink service status:
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation HD Graphics 630 (rev 04)
NVIDIA Corporation GP106M [GeForce GTX 1060] (rev a1)
VGA (3D):
X11 version: 1.20.4-1+deb10u3
X11 configs: /etc/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf.d/synaptics.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

Barabazs, 2021-07-07:
Hi @ramifications123 
The log indicates that the system is not using `systemd`. 
> System has not been booted with systemd as init system (PID 1). Can't operate.

ramifications123, 2021-07-07:
Hi, I used systemd and it doesn't recognize the monitor. This is what I got for the debug.

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-17-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation HD Graphics 630 (rev 04)
NVIDIA Corporation GP106M [GeForce GTX 1060] (rev a1)
VGA (3D):
X11 version: 1.20.4-1+deb10u3
X11 configs: /etc/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf.d/synaptics.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel

-------------------------------------------------------------------

Barabazs, 2021-07-08:
Ok that looks better. Please try this config:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

ramifications123, 2021-07-09:
How would I update the config? I saw the OutputClass in the shell script. Which file should I update?

Barabazs, 2021-07-15:
@ramifications123 replace the content of this file `/etc/X11/xorg.conf.d/20-displaylink.conf` with the config in my previous post.

ramifications123, 2021-07-15:
I replaced the section with your previous post and it crashed my display/monitor. 
I placed my Section "Device" back and put the "OutputClass" below it. It crashed it as well 

It works well if I don't mess with the original settings.

closed 2021-12-21 by Barabazs