AdnanHodzic/displaylink-debian #590: Can't work on UOS system

Eddie-064, 2021-07-04:
* Are you using the latest driver? yes
    
  * Are you using the latest EVDI version? 1.9.1 release
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website? yes
    
  * Is this issue related to evdi/kernel? yes
    

  * Linux distribution and its version UOS 20
  * Linux kernel version 5.4.50-amd64-desktop (deepin@deepin-PC)
  * Xorg version (if used)  1.20.4
  * Desktop environment in use 
  * eddie@eddie-PC:~/Desktop$ echo $XDG_CURRENT_DESKTOP
Deepin

after plug in displaylink device , the screen show nothing.

when plug in displaylink device the debug message show blow:
eddie@eddie-PC:~/Desktop$ dmesg | grep evdi
[ 4.443782] evdi: [I] Initialising logging on level 5
[ 4.443784] evdi: [I] Atomic driver: yes
[ 70.413212] evdi: [D] evdi_crtc_init:413 drm_crtc_init: 0 p00000000154fb118
[ 70.413231] evdi: [D] evdi_detect:95 (dev=0) poll connector state: disconnected
[ 70.413256] evdi: [W] evdi_painter_send_update_ready_if_needed:706 Painter does not exist!
[ 70.413704] [drm] Initialized evdi 1.9.1 20210126 for evdi.0 on minor 0
[ 70.413726] evdi: [I] Evdi platform_device create
[ 70.413730] evdi: [I] Attaching to usb:1-4.1.1
[ 70.416037] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[ 70.416039] evdi: [I] Task 1412 (Xorg) of process 1412 (Xorg)
[ 70.416774] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[ 70.416776] evdi: [I] Task 1412 (Xorg) of process 1412 (Xorg)
[ 70.417137] evdi: [D] evdi_detect:95 (dev=0) poll connector state: disconnected
[ 70.417255] evdi: [D] evdi_painter_framebuffer_size:594 Scanout buffer not set.
[ 70.417257] evdi: [D] evdi_painter_mark_dirty:622 (dev=0) Skip clip rect. Scanout buffer not set.
[ 70.417271] evdi: [D] evdi_detect:95 (dev=0) poll connector state: disconnected
[ 70.419427] evdi: [D] evdi_painter_dpms_notify:716 (dev=0) Notifying dpms mode: 3
[ 70.419430] evdi: [W] evdi_painter_send_event:321 Painter is not connected!
[ 70.430927] evdi: [D] evdi_painter_dpms_notify:716 (dev=0) Notifying dpms mode: 3
[ 70.430930] evdi: [W] evdi_painter_send_event:321 Painter is not connected!
[ 70.437716] evdi: [D] evdi_detect:95 (dev=0) poll connector state: disconnected
[ 70.437742] evdi: [D] evdi_detect:95 (dev=0) poll connector state: disconnected
[ 70.446502] evdi: [D] evdi_painter_connect:852 (dev=0) Process is trying to connect
[ 70.446506] evdi: [I] Task 1463 (DesktopManagerE) of process 1431 (DisplayLinkMana)
[ 70.446669] evdi: [D] evdi_add_i2c_adapter:814 (dev=0) Added i2c adapter bus number 0
[ 70.446674] evdi: [D] evdi_painter_connect:904 (dev=0) Connected with 00000000e627f9df
[ 70.446678] evdi: [D] evdi_detect:91 (dev=0) poll connector state: connected
[ 70.452244] evdi: [D] evdi_detect:91 (dev=0) poll connector state: connected
[ 70.452248] evdi: [D] evdi_painter_get_edid_copy:242 (dev=0) EDID valid
[ 70.452303] evdi: [D] evdi_detect:91 (dev=0) poll connector state: connected
[ 70.452304] evdi: [D] evdi_painter_get_edid_copy:242 (dev=0) EDID valid
[ 71.439841] evdi: [D] evdi_detect:91 (dev=0) poll connector state: connected
[ 71.439847] evdi: [D] evdi_painter_get_edid_copy:242 (dev=0) EDID valid
[ 71.439906] evdi: [D] evdi_detect:91 (dev=0) poll connector state: connected
[ 71.439908] evdi: [D] evdi_painter_get_edid_copy:242 (dev=0) EDID valid

and after reboot the system will get stuck, and can't get into desktop.
and I think the problem is "/etc/modprobe.d/evdi.conf" file, because after I remove it, I can success get into desktop but still not wok on displaylink device.
I try get into tty to show the following message: ( just reboot and not plug any displaylink device)
[image]

and the displaylink-debian debug show below:( the problem is same on the intel cpu and intel gpu )
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: d

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Uos
Release: n/a
Kernel: 5.4.50-amd64-desktop

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: 
Subsystem: GPU
Definition
VGA: Zhaoxin ZX-E C-960 GPU
VGA (3D): 
X11 version: 1.20.4.9-1+dde
X11 configs: /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/xorg.conf.d/75-wacom.conf /etc/X11/xorg.conf.d/40-libinput.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Eddie-064, 2021-07-07:
since displaylink-debian.sh is not support for UOS, so I modify the shell if-else to true at line 154.
so displaylink-debian.sh could run all script by install.

Barabazs, 2021-07-08:
I edited your first post to make the logs more readable. 

What is in the evdi.conf file?

Eddie-064, 2021-07-08:
@Barabazs  the content is 
options evdi initial_device_count=4

Barabazs, 2021-07-27:
Is secure boot disabled?

closed 2021-12-21 by Barabazs