AdnanHodzic/displaylink-debian #591: debian testing fails to boot after installation

aelray, 2021-07-05:
Nothing suspicious in dmesg except for several of those
```
Jul  5 20:30:41 thinkpadL15 kernel: [    2.442513] evdi: [D] evdi_driver_postclose:230 (dev=0) Process tries to close us, postclose
Jul  5 20:30:41 thinkpadL15 kernel: [    2.442513] evdi: [I] Task 868 (Xorg) of process 868 (Xorg)
Jul  5 20:30:41 thinkpadL15 kernel: [    2.442516] evdi: [D] evdi_driver_postclose:230 (dev=0) Process tries to close us, postclose
Jul  5 20:30:41 thinkpadL15 kernel: [    2.442516] evdi: [I] Task 868 (Xorg) of process 868 (Xorg)
```
output of  `sudo ./displaylink-debian --debug`:
```
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-7-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: 
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne (rev d1)
VGA (3D): 
X11 version: 1.20.11-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

```

Barabazs, 2021-07-07:
Hi @aelray . The information you provided is insufficient to help you with this issue.
Please post the install log, debug log (after it's installed) and boot log.

aelray, 2021-07-08:
Hi,
attached is the install log and the boot log. Debug log is in my original post, right?
[install.log](https://github.com/AdnanHodzic/displaylink-debian/files/6785933/install.log)
[boot.log](https://github.com/AdnanHodzic/displaylink-debian/files/6785938/boot.log)

Eddie-064, 2021-07-12:
HI  @Barabazs 
I have test this problem,it can boot success after remove /etc/modprobe.d/evdi.conf.
but displaylink moniter can not work.

Barabazs, 2021-07-27:
Is Secure Boot disabled?

aelray, 2021-07-27:
Secure Boot was/is disabled. Anyhow my dock doesn't use DisplayLink so in the end I didn't need displaylink-debian. I just had problems with my dock because I hadn't installed AMD firmware (which Debian of course doesn't do by default).

closed 2021-08-19 by Barabazs