AdnanHodzic/displaylink-debian #592: Support for Pop!_OS 21.04 Hirsute

bergben, 2021-07-08:
Would be nice to get support for Pop!_OS 21.04 Hirsute.

Can I test by aadding it to https://github.com/AdnanHodzic/displaylink-debian/blob/c5cba0e75f4b99abdfa1efc5ee195c66b707a5d7/displaylink-debian.sh#L258 ?

--------------- Linux system info ----------------

Distro: Pop
Release: hirsute
Kernel: 5.11.0-7620-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 630 (Mobile)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Barabazs, 2021-07-08:
Hi @bergben 

You're on the right track. Just modify the line to
`	if [ $codename == "focal" ] || [ $codename == "groovy" ] || [ $codename == "hirsute" ] || [ $codename == "n/a" ];`

Let me know how it goes.

hmmmmniek, 2021-07-11:
i tried this on popos 21.04. install went fine! But the display refreshrate (on both my laptop display and the external display) is very slow whenever something moves on either screen.. Is this a known issue? Anyone know how to fix this?

develm, 2021-07-13:
Hi,

I just installed DisplayLink driver on Pop OS 21.04 on kernel 5.12.16 and it works perfectly

PR #595 (2021-07-15): Add support for Pop OS 21.04 hirsute
closes #592
closed 2021-07-15 by Barabazs