AdnanHodzic/displaylink-debian #594: failed to load driver: evdi / MESA-LOADER: failed to open evdi

2ZZ, 2021-07-14:
* Linux distribution and its version:  Ubuntu 18.04 LTS
  * Linux kernel version: 4.15.0-147-generic and 5.4.0-58-generic
  * Xorg version: 1.20.8
  * Desktop environment in use: None
  * Screen hardware:  DisplayLink PA35 USB Display

Hi

Any help with the error below would be greatly appreciated, this is the message we receive when trying to start Xorg with the evdi driver installed from displaylink-debian
```
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Jul 14 10:24:11 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
MESA-LOADER: failed to open evdi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: evdi
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open evdi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: evdi
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open evdi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: evdi
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open evdi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: evdi
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
(II) modeset(G0): Initializing kms color map for depth 24, 8 bpc.
(II) modeset(G1): Initializing kms color map for depth 24, 8 bpc.
(II) modeset(G2): Initializing kms color map for depth 24, 8 bpc.
(II) modeset(G3): Initializing kms color map for depth 24, 8 bpc.
randr: falling back to unsynchronized pixmap sharing
```
The screen itself shows a garbled image, see: https://i.ibb.co/68RGBcv/20210714-111810.jpg

This is the contents of 20-displaylink.conf, we also tried the one created by the installer.
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
Section "Device"
  Identifier "Intel Graphics"
  Driver "intel"
EndSection
```
The evdi module is loaded
```
root@1d9bbc2ad6b6:~# modinfo evdi
filename:       /lib/modules/4.15.0-147-generic/updates/dkms/evdi.ko
license:        GPL
description:    Extensible Virtual Display Interface
author:         DisplayLink (UK) Ltd.
srcversion:     E62D7CF65F9E0EDBA4080B0
depends:        drm,drm_kms_helper,syscopyarea,sysfillrect,sysimgblt
retpoline:      Y
name:           evdi
vermagic:       4.15.0-147-generic SMP mod_unload modversions
parm:           initial_loglevel:Initial log level (int)
parm:           initial_device_count:Initial DRM device count (default: 0) (ushort)
``` 
We are enabling the screen with xrandr --auto
```
xrandr --output DVI-I-1-1 --auto
```
And it looks to be configured with the correct resolution according to the manufacturer specs
```
root@1d9bbc2ad6b6:~# xrandr --current
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 8192 x 8192
DP-1 disconnected primary (normal left inverted right x axis y axis)
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-2 connected 1920x1080+0+0 (normal left inverted right x axis y axis) 476mm x 268mm
   1920x1080     62.58*+
HDMI-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 connected 1366x768+0+0 (normal left inverted right x axis y axis) 223mm x 126mm
   1366x768      60.00*+
   ```
   These are the providers
   ```
   root@1d9bbc2ad6b6:~# xrandr --listproviders
Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0xe1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xbf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x9d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x7a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```
We have also tried directly using the manufacturer's installer with similar results, each time uninstalling, unplugging the screen & rebooting for a clean install. We also tried 3 different kernel versions.
- displaylink-driver-5.2.14.run
- 'DisplayLink USB Graphics Software for Ubuntu5.2-EXE.zip'
- 'DisplayLink USB Graphics Software for Ubuntu5.4-EXE.zip'

The screen itself works ok in Windows.

Thanks

2ZZ, 2021-07-14:
I am also trying to confirm if we should be using udl or evdi as there is some conflicting information. This is the device as detected on boot
```
[    3.842795] usb 1-4.1.1: new high-speed USB device number 10 using xhci_hcd
[    3.947506] usb 1-4.1.1: New USB device found, idVendor=17e9, idProduct=039a, bcdDevice= 2.20
[    3.947508] usb 1-4.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.947509] usb 1-4.1.1: Product: PA35 USB Display
[    3.947510] usb 1-4.1.1: Manufacturer: DisplayLink
[    3.947510] usb 1-4.1.1: SerialNumber: 490056
[    3.952707] [drm] vendor descriptor length:17 data:17 5f 01 00 15 05 00 01 03 00 04
[    4.022742] usb 1-4.4: new full-speed USB device number 11 using xhci_hcd
[    4.045065] fbcon: Deferring console take-over
[    4.045068] udl 1-4.1.1:1.0: fb1: udldrmfb frame buffer device
[    4.045206] [drm] Initialized udl 0.0.1 20120220 for 1-4.1.1:1.0 on minor 1
[    4.045207] [drm] Initialized udl on minor 1
[    4.045236] usbcore: registered new interface driver udl
```
lshw shows
```
                    *-usb:0
                         description: Generic USB device
                         product: PA35 USB Display
                         vendor: DisplayLink
                         physical id: 1
                         bus info: usb@1:4.1.1
                         version: 2.20
                         serial: 490056
                         capabilities: usb-2.00
                         configuration: driver=usbfs maxpower=500mA speed=480Mbit/s
```
But either way both udl and evdi are not displaying correctly. 
The screen is showing a green band down the middle in some configurations: https://i.ibb.co/hRh7mwS/20210714-143555.jpg

2ZZ, 2021-07-14:
Adding debug info
```
root@1d9bbc2ad6b6:~/displaylink-debian# ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.4.0-77-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D):
X11 version: 1.20.8-2ubuntu2.2~18.04.5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
  Identifier "Intel Graphics"
  Driver "modesetting"
EndSection

Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0xe1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xbf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x9d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x7a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

The screen corruption seems to be related to this message from Xorg
```
randr: falling back to unsynchronized pixmap sharing
```

Barabazs, 2021-07-27:
Could you try to replace your config with this one:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

closed 2021-08-19 by Barabazs