AdnanHodzic/displaylink-debian #599: Disable modesetting

lucaorioli, 2021-07-25:
Hi, I need to run the driver adding nomodeset option in kernel (which disable modesetting).

Is it possible?
How can I do it?

Thank you very much

Barabazs, 2021-07-27:
Hi @lucaorioli. Your question is unrelated to the use of displaylink-debian. A simple Google search should give you the answer you need.

lucaorioli, 2021-07-27:
I haven't explauined well, sorry. 
I'm asking how to disable modesetting in displaylink driver, if I disable it at kernel level (adding nomodeset kernel parameter) displaylink doesn't work. But I need to disable modesetting or my integrated screen won't work

> Hi @lucaorioli. Your question is unrelated to the use of displaylink-debian. A simple Google search should give you the answer you need.

Barabazs, 2021-07-28:
Please post the output of `sudo ./displaylink-debian.sh --debug`

lucaorioli, 2021-08-01:
With Modesetting enabled:

--------------- Linux system info ----------------

Distro: Ubuntu
Release: hirsute
Kernel: 5.11.0-26-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Core
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x154 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x132 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x110 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xee cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

lucaorioli, 2021-08-01:
If I disable modesetting (adding "nomodeset" option to kernel) the system remain in console(gdm can't start) and  
sudo ./displaylink-debian.sh --debug
returns: No display.
To let gdm start I need to uninstall displaylink driver.
I need displaylink driver works with nomodeset kernel option

Barabazs, 2021-08-19:
I don't think that is possible without using nomodeset.

Your question is still unrelated to `displaylink-debian` though.
Try asking your question on askubuntu.com or on the forums.

closed 2021-08-19 by Barabazs

lucaorioli, 2021-08-19:
Why Are you continuing to say it isn't related? I need displaylink works
without nomodeset! Ubuntu works without nomodeset, displaylink doesn't
works without nomodeset, so it's related to displaying, not related to
Ubuntu!

Please help me!