AdnanHodzic/displaylink-debian #600: Process tries to close us, postclose

sfuhrm, 2021-07-27:
On Debian 10 with 5.10.0-0.bpo.7-amd64 kernel, I am not able to use this.

After successful installation and reboot, I'm getting a login (gdm3) screen. So far so good.
As soon as I want to login, I get a console screen saying (excerpt from dmesg):

{noformat}
...
[  389.961502] evdi: [I] Task 4112 (Xorg.wrap) of process 4112 (Xorg.wrap)
[  389.961517] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  389.961520] evdi: [I] Task 4112 (Xorg.wrap) of process 4112 (Xorg.wrap)
[  389.961533] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  389.961571] evdi: [I] Task 4112 (Xorg.wrap) of process 4112 (Xorg.wrap)
[  389.961592] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  389.961595] evdi: [I] Task 4112 (Xorg.wrap) of process 4112 (Xorg.wrap)
[  390.027529] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  390.027591] evdi: [I] Task 4112 (Xorg) of process 4112 (Xorg)
[  390.027645] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  390.027695] evdi: [I] Task 4112 (Xorg) of process 4112 (Xorg)
[  390.027736] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  390.027786] evdi: [I] Task 4112 (Xorg) of process 4112 (Xorg)
[  390.027826] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  390.027875] evdi: [I] Task 4112 (Xorg) of process 4112 (Xorg)
[  406.036123] nouveau 0000:01:00.0: sec2: unhandled intr 00000010
[  435.110578] nouveau 0000:01:00.0: sec2: unhandled intr 00000010
[  436.305789] nouveau 0000:01:00.0: sec2: cmdq: timeout waiting for queue ready
[  436.305801] nouveau 0000:01:00.0: gr: init failed, -110
[  441.406177] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  441.406180] evdi: [I] Task 4373 (Xorg.wrap) of process 4373 (Xorg.wrap)
[  441.406186] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  441.406187] evdi: [I] Task 4373 (Xorg.wrap) of process 4373 (Xorg.wrap)
[  441.406192] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  441.406193] evdi: [I] Task 4373 (Xorg.wrap) of process 4373 (Xorg.wrap)
[  441.406198] evdi: [D] evdi_driver_postclose:231 (dev=0) Process tries to close us, postclose
[  441.406199] evdi: [I] Task 4373 (Xorg.wrap) of process 4373 (Xorg.wrap)
{noformat}

After that, X11 is returned to the login screen.
Can you assist?

Barabazs, 2021-07-27:
Please post the output of `sudo ./displaylink-debian.sh --debug`

sfuhrm, 2021-07-27:
Thanks for the hint! I moved away /etc/X11/xorg.conf.d/20-displaylink.conf, then a login was possible again. I have a intel gfx card, so I didn't know that this file needs to be moved away.

After

```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0
```

it worked!

Here's the debug output anyway:

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.10.0-0.bpo.7-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): NVIDIA Corporation Device 1c90 (rev a1)
X11 version: 1.20.4-1+deb10u3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Datei oder Verzeichnis nicht gefunden
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0xf0 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0xc7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xa5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x83 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x61 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Maybe it helps?

Barabazs, 2021-07-28:
Hi @sfuhrm. I'm glad to hear it's working for you.

closed 2021-07-28 by Barabazs