AdnanHodzic/displaylink-debian #603: HP DV6 - having dual graphics was interchangeable when still using Windows 7

kurtborg92, 2021-08-04:
Hi I am trying to use a Targus USB3 Universal Dock, I got the usb and ethernet to work but the display is still not being displayed to date maybe you can help me with this. I tried all of the solutions you gave in the post-installation and even removed the conf file and re-created it again using different settings but it is still not working.

Any help will be greatly appreciated.

Thanks,
Kurt

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-17-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Processor
Radeon
VGA: Intel Corporation Core Processor Integrated Graphics Controller (rev 02)
Advanced Micro Devices, Inc. [AMD/ATI] Madison [Mobility Radeon HD 5650/5750 / 6530M/6550M]
VGA (3D): 
X11 version: 1.20.4-1+deb10u3
X11 configs: /etc/X11/xorg.conf.d/synaptics.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/10-evdev.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x43 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 2 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Barabazs, 2021-09-13:
Please try to reinstall and post the output of the installer, and then also the debug output.

kurtborg92, 2021-09-16:
Hi Barabazs,

Re-Installer Output:

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: r

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-17-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

DisplayLink Linux Software 5.4.0-55.153 install script called: uninstall
Distribution discovered: Debian GNU/Linux 10 (buster)

Uninstalling

[ Removing EVDI from kernel tree, DKMS, and removing sources. ]
dkms remove evdi/1.9.1 --all

-------- Uninstall Beginning --------
Module:  evdi
Version: 1.9.1
Kernel:  4.19.0-17-amd64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

evdi.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.0-17-amd64/kernel/drivers/gpu/drm/evdi/
rmdir: failed to remove directory 'kernel/drivers/gpu/drm': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

[... 214 lines cut ...]

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Need anything else let me know.

Thanks,
Kurt

Barabazs, 2021-09-17:
What is the content of these files:
 /etc/X11/xorg.conf.d/synaptics.conf
/etc/X11/xorg.conf.d/10-evdev.conf 
/etc/X11/xorg.conf

Maybe they are causing a conflict.

closed 2021-12-21 by Barabazs