AdnanHodzic/displaylink-debian #604: Pop! OS 21.04 hirsute ASUS MB169C+ not detected

jfsantos-ds, 2021-08-14:
I followed the installation guide and tried the troubleshooting suggestions (deleting X11 conf file and reboot).
Did a successful reinstall but the problem persists, I haven't managed to connect the new display yet.

Distribution version:
```
$ cat /etc/lsb-release
DISTRIB_ID=Pop
DISTRIB_RELEASE=21.04
DISTRIB_CODENAME=hirsute
DISTRIB_DESCRIPTION="Pop!_OS 21.04"
```

Debug output:
```
$ sudo ./displaylink-debian.sh --debug
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Pop
Release: hirsute
Kernel: 5.11.0-7620-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation CometLake-H GT2 [UHD Graphics] (rev 05)
NVIDIA Corporation TU104M [GeForce RTX 2070 SUPER Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x30c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2ea cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2c8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e3 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

What should I try next?

Barabazs, 2021-08-19:
Can you reinstall and then repost the debug output please?

Those monitors are notoriously a PITA on linux...

jfsantos-ds, 2021-08-19:
Hey @Barabazs ! Thank you for coming to my aid, will jump right on it. BRB with your request.

jfsantos-ds, 2021-08-19:
1- Unninstalled (done)
2- Re-Installation:
```(base) fsantos@pop-os:~/GitRepos/displaylink-debian$ sudo ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-7620-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/3751?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2021-08-19 16:42:20--  https://www.synaptics.com/sites/default/files/exe_files/2021-04/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 34.210.13.121, 52.89.62.198
Connecting to www.synaptics.com (www.synaptics.com)|34.210.13.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14501284 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.zip’

DisplayLink_Ubuntu_5.4. 100%[=============================>]  13.83M  3.55MB/s    in 4.3s    

2021-08-19 16:42:25 (3.23 MB/s) - ‘DisplayLink_Ubuntu_5.4.zip’ saved [14501284/14501284]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.zip
  inflating: 5.4/displaylink-driver-5.4.0-55.153.run  
Creating directory displaylink-driver-5.4.0-55.153
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.0-55.153  100%  
patching file 5.4/displaylink-driver-5.4/displaylink-installer.sh
Hunk #1 succeeded at 456 (offset -1 lines).

Installing driver version: 5.4

DisplayLink Linux Software 5.4.0-55.153 install script called: install
Distribution discovered: Pop!_OS 21.04
WARNING: Version 1.9.1 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.
```
3- debug output
```
(base) fsantos@pop-os:~/GitRepos/displaylink-debian$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Pop
Release: hirsute
Kernel: 5.11.0-7620-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: 
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation CometLake-H GT2 [UHD Graphics] (rev 05)
NVIDIA Corporation TU104M [GeForce RTX 2070 SUPER Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x316 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2f4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2d2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2b0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1fb cap: 0x2, Sink Output crtcs: 3 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

jfsantos-ds, 2021-08-23:
Closing the issue.
My apologies for raising it but it seems that my laptop MSI GL65 SLSK does not support video output through the USB-C port. First I have a connectivity problem to solve.

closed 2021-08-23 by jfsantos-ds