AdnanHodzic/displaylink-debian #609: Can't get second display to work under Debian Buster via displaylink. It works under Ubuntu 20

assembleer, 2021-08-22:
Hello,
Can't get external display to work under Debian Buster with displaylink. With Ubuntu 20 it works, so it's not a hardware thing. Maybe it's driver, as in Ubuntu it has a oibaf driver (Tuxedo Polaris 15 gen 1 with Nvidia rtx2060 mobile videocard). Currently I'm using buster backports to get the videocard working properly under Debian. Any advice is welcome.

The output from ./displaylink-debian.sh --debug
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.10.0-0.bpo.8-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: nvidia
amdgpu
Subsystem: TU106M
Renoir
VGA: NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c6)
VGA (3D): 
X11 version: 1.20.4-1+deb10u3
X11 configs: /etc/X11/pk20210808_xorg.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Bestand of map bestaat niet
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x44 cap: 0x0 crtcs: 4 outputs: 1 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:05:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2021-09-13:
How did you install the driver?

assembleer, 2021-09-13:
At first with the stable kernel (4) in Buster I couldn’t get it to work.

However after installing the backport kernel (5j I succeeded in getting it to work.
Recently I upgraded to Debian Bullseye and with the new kernel delivered with stable it now works properly.

Thx for your response and willingness to help.

Peter

Barabazs, 2021-09-13:
OK thank you for your feedback. I'm glad it worked out for you.

closed 2021-09-13 by Barabazs