AdnanHodzic/displaylink-debian #613: Freeze of Pop OS Gnome UI when adjusting display settings

bwyyoung, 2021-09-05:
After installing displaylink debian, on POP! OS 21.04,

All monitors show up as displaylink.
I have 1 laptop screen - 1920x1080
External Displaylink screens:
2560x1440
1920x1080

When I go under display settings in POP! OS and try to rearrange the positions of the monitors, and click APPLY,
the whole UI freezes for POP! OS.

Pressing CTRL ALT F1 / F2 does not work, whole system hang.

This has occurred on both official driver, and this custom driver in github.

Any idea on the cause of the problem?

RaphaelJenni, 2021-09-09:
Same problem for me on Ubuntu.

The-Almighty-Sasquatch, 2021-09-11:
I too have this issue, it has been long running and I have posted several issues. Is there a work around for this at all?

Thanks

Barabazs, 2021-09-13:
Unfortunately nothing that displaylink-debian can do. It's probably settings related and requires fine-tuning.

closed 2021-09-13 by Barabazs

SamSpiri, 2021-09-25:
As workaround I manually entered desired settings into my ~/.config/monitors.xml and never touched settings in GUI