AdnanHodzic/displaylink-debian #614: Kali rolling-release install

area51pilot, 2021-09-07:
I receive the following error at the end of an attempted install:, what can I do to correct this? 

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.1.zip
  inflating: 5.4.1/displaylink-driver-5.4.1-55.174.run  
chmod: cannot access '5.4.1/displaylink-driver-5.4.1.[0-9]*.run': No such file or directory

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.10.0-kali9-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Iris
VGA: Intel Corporation Iris Plus Graphics G7 (rev 07)
VGA (3D): 
X11 version: 1.20.11-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

ApeironTsuka, 2021-09-09:
Judging by the error, this is caused by a change in upstream file naming ("5.4.1-55.174" the extra . in there isn't expected by the script). As a work around for now, try changing `[0-9]*` to `[0-9.]*` in lines 423, 424, and 425 of displaylink-debian.sh. There's probably a better way to change the regex for it but this should be good enough to get it working in the meantime.

ApeironTsuka, 2021-09-09:
Oh, oops. This is what I get for responding to this on almost no sleep. I didn't notice the new filename had a - in it. Change those same 3 lines to have `[.-][0-9.]*` instead. Sorry about that.

area51pilot, 2021-09-09:
Thank you for the help ... I hope you got some rest.
I now have this error:
Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.1.zip
  inflating: 5.4.1/displaylink-driver-5.4.1-55.174.run  
chmod: invalid option -- ']'
Try 'chmod --help' for more information.
                                                                                
┌──(area51pilot㉿kali)-[~/displaylink-debian]
└─$           

Heres the script as edited so far: https://pastebin.com/hiNXC6qX

ApeironTsuka, 2021-09-10:
Hmmm. It should look like `displaylink-driver-${version}[.-][0-9.]*.run` (except for line 425 where `.run` is absent).

area51pilot, 2021-09-10:
so here is something interesting ... it seems to somewhat work now.

command: 

xrandr --listproviders

output: 

Providers: number: 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

so I set the following:

xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0

... and 1 out of 2 external displays works (this Dell D3100 dock works normally in Ubuntu 21.x)
So I'm close but now trying to figure out the second display.  Both connectors are HDMI dock <->monitor
If I unplug the monitors and plug them in using reverse order (not the port just the process of connecting one at a time to the monitor, the second monitor works, and the first doesn't (dock cabling left untouched)?

All devices have been restarted (laptop, dock, monitor power cycle) ... not sure where to go from here?

area51pilot, 2021-09-10:
OK ... so here is something else thats really weird, just closed the lid to undock and connect my Ubuntu laptop and both external monitors came on? then I opened the lid and now I have all 3???

... so I guess its working ( I think im working more though to use this distro on a dock  :P  )

... thank you for all of your help.

ApeironTsuka, 2021-09-10:
Them being off/black when initially plugged in is apparently an [upstream known issue](https://support.displaylink.com/knowledgebase/articles/641668-known-issues-with-displaylink-ubuntu-support) and you should be able to go into system settings -> displays and turn them on if/when it happens. They probably all turn on when you open the main panel or Ubuntu turns them on by default if the main panel is off. Not really sure on that - I don't (currently) have a DisplayLink screen nor an Ubuntu machine.

Barabazs, 2021-09-13:
Thank you for reporting this. The script has been updated.

closed 2021-09-13 by Barabazs