AdnanHodzic/displaylink-debian #619: DisplayLink on Debian 11 / Asus UX430U (Intel UHD620+nVidia GP108M/GeForce MX150)

MrBig1964, 2021-09-13:
I've used your installation procedure to try to get my DisplayLink device working in Debian 11.
The only way I achieve to get KDE running is deleting (or renaming) the 20-displaylink.conf file.
But the system insists on not recognizing the DisplayLink which is connected and doesn't show any other display apart the laptop main one.
The debug info I get from running displaylink-debian is:

--------------- Linux system info ----------------
Distro: Debian
Release: bullseye
Kernel: 5.10.0-8-amd64

---------------- DisplayLink info ----------------
Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------
Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
X11 version: 1.20.11-1

-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 
-------------------- Monitors --------------------
Providers: number : 6
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x16a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x148 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x126 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x104 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0xd2 cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

I've already tried several different suggestions like using xrandr setprovideroutputsource to assign providers 1 to 4 to 0, but it was a no go. Also using nomodeset at boot does anything usefull.
I'm not a Linux or Debian expert and I could use some help to try to tracking down this issue.
Could someone help me?
Thanks in advance.

Barabazs, 2021-09-13:
Can you confirm that I understand this correctly please?
Your computer does not boot successfully when the file "20-displaylink.conf" exists?

I see that providers are detected, so maybe you should try to associate them as noted in the [post-install-guide#display-detection](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection).

MrBig1964, 2021-09-13:
If the file 20-displaylink.conf exists, the computer boots but don't go to the graphic interface.
If I do ALT-F4 and go to a TTY, I can log-in to the shell using my usual username / password combination.
I've tried also to associate the displaylink providers (1 to 4) to the graphic card (0). Nothing happens...

MrBig1964, 2021-09-14:
By the way, just for information, the standard DisplayLink driver works with no issues in Ubuntu 20.04.

MrBig1964, 2021-09-16:
New information. When I do:

   xrandr --setprovideroutputsource 4 0

and after this,

   sudo systemctl restart displaylink-driver.service

I get my external screen to be detected in the KDE control panel.
But even saying that is enabled, the screen keeps staying black and doesn't work.
I think I must very close to make this thing work but it is still a no go.
Any more suggestions?

Barabazs, 2021-09-17:
> By the way, just for information, the standard DisplayLink driver works with no issues in Ubuntu 20.04.

Correct. It's the only officially supported platform.

> New information. When I do:
>
> xrandr --setprovideroutputsource 4 0
>
> and after this,
>
> sudo systemctl restart displaylink-driver.service
>
> I get my external screen to be detected in the KDE control [...]

Just to clarify, all providers should be associated. Did you only try to associate 4 to 0?

MrBig1964, 2021-09-17:
> Just to clarify, all providers should be associated. Did you only try to associate 4 to 0?
Just because is the only one that apparently does something.
When I associate all others, nothing changes.
Perhaps because this one is the one corresponding with the physical video port to where my monitor is connected.

MrBig1964, 2021-09-18:
> Just to clarify, all providers should be associated. Did you only try to associate 4 to 0?

No. I've associated all available, from 1 to 4. Then I have restarted the displaylink service.
The screen was detected but no image.

closed 2021-12-21 by Barabazs