AdnanHodzic/displaylink-debian #621: slow/lagging video/display on vega10/ryzen 7

dkebler, 2021-09-16:
opened this but it is probably realted to #265 

I had been using my displaylink device for couple years with an intel I5 (integrated gpu) without issue.
Just switched over to a ryzen 7 3750u machine

The issue is that the display is lagging/slow which is very apparent when playing a video (jerky) but one can tell the mouse is also lagging so it's some refresh rate issue(?) and not specific to steaming video playback.  When I unplug the displaylink device it goes away, plug back in it returns.

this default conf file (set by install) is 

```
Section "OutputClass"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option  "AccelMethod" "none"
EndSection
```

and I have also tried with no conf file but the issue remains.

I run linux mint 20 and it's not possible to use wayland (like suggested in issue #265) so that is not a solution for me.
maybe I need accelleration of some kind or a different conf file

Any suggestions. Anyone having similar issue with veg10/ryzen embeeded video and solved it?

here is my debug output.  That's mint 20 which uses focal 20.04.
I have a pluggable dual hdmi/ethernet displaylink device, usbc-6950-hdmi

```
--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana  (focal 20.04)
Kernel: 5.4.0-84-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
1.9.1
DisplayLink service status: 
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: amdgpu
i2c_amd_mp2
Subsystem: Inc.
Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 
note: default mentioned above did not work either

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x57 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 4 name:AMD Radeon(TM) Vega 10 Graphics @ pci:0000:05:00.0
Provider 1: id: 0x100 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xde cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xbc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x9a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

dkebler, 2021-09-16:
I install amdgpu driver for vega 10
https://amdgpu-install.readthedocs.io/en/latest/install-script.html
and reinstalled displaylink-debian.

issue persists. :(

Barabazs, 2021-09-17:
That looks like the same problem others are experiencing on admgpu.
It's still not fixed as far as I can tell: https://github.com/DisplayLink/evdi/issues/271

I'll close this issue as this is not related to displaylink-debian.

closed 2021-09-17 by Barabazs

dkebler, 2021-09-22:
Reporting that I built the evdi driver from the branch/fork (mentioned at that issue, link below) and replaced the one built via this script and this seems to have resolved the issue for now without consequences.  I'm using with NO xorg conf file.  My understanding is the evdi driver will eventaully get a bonfide bug fix for this amd gpu specific issue.

https://github.com/pioto1225/evdi/tree/amd_vmap_texture

dkebler, 2021-11-09:
Reporting that the instructions for installing that fork have been updated and now dkms to install if you manually update (per link at fork) dkms as focal universe version is way behind

dkms will make it so the driver is brought forward with each kernel update.  I've been using this hack/fix for several months and it doesn't cause crashes for me.  It has been merged into the dev branch at the main repo but not into master.