AdnanHodzic/displaylink-debian #622: DisplayLink stopped working after apt upgrade

AeolusTech, 2021-09-18:
I have Secure Boot enabled as this is company requirement - it worked before flawlessly and I have never experienced any issues. I have disabled it but it did not help at all. I have used both the scripts provided in this repo, as well as official [Display Link software](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu-5.4.1?filetype=exe). However without any luck.

I have tried [this workaround](https://wiki.archlinux.org/title/DisplayLink#Workaround_1:_Use_older_intel_driver_as_a_fallback) as my Xorg is version 1.20.11.

$ dpkg -l |grep xserver-xorg-core
ii  xserver-xorg-core                          2:1.20.11-1ubuntu1~20.04.2                amd64        Xorg X server - core server

I would greatly appreciate Your help as this problem is so frustrating for me...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.11.0-34-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4a cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 7 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

calebanthony, 2021-09-21:
It also stopped working for me with a recent `apt update` (I'm on PopOS 21.04). Here's the steps I used to get back up and running:

1. Reboot (if you have been making any changes to the driver).
2. Get an updated version of the repo by cloning, then uninstall.
3. Reboot
4. Install
5. Reboot
6. Update the xorg conf file via [this comment](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348).
7. Reboot

Barabazs, 2021-09-30:
@AeolusTech Can you confirm that the procedure from @calebanthony worked?

AeolusTech, 2021-09-30:
Hi guys,
the issue persists on my work PC - and recently I have been stuffed with work. I haven't tried that solution yet. Gimme couple more days, please :)

AeolusTech, 2021-09-30:
Ok, so at step 4. the installation freezes my PC, and the following artifacts appear on my Desktop. I will try to update the xorg conf and reboot and get back to you

[image]

The installer causes a current tty desktop freeze just after printing line "Creating symlink to displaylink-driver.service". I am able to switch to another tty (tty6) and reboot the PC from that place

AeolusTech, 2021-09-30:
Yup, it didn't work. I will repeat the process one more time, but this time I will make sure that after uninstallation the file /etc/X11/xorg.conf.d/20-displaylink.conf no longer exists

AeolusTech, 2021-09-30:
I have also unplugged the HDMI cable.

AeolusTech, 2021-09-30:
I've managed to do an OCR from my phone when Desktop froze

kke@SS-GCCZL13:~/Downloads/displaylink-deblan
Q

cp libevdi.so.1.9.1 libevdi.so
[ Installing x64-ubuntu-1604/DisplayLinkManager]
[ Installing libraries ]
[ Installing firmware packages ]
Installing licence file ]
Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager scripts ]
Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubunt
u
Installation complete!
Adding udlfb to blacklist
Adding udl to blacklist
Performing post install steps
I
Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.s
ervice
- /lib/systemd/system/displaylink-driver.service.

AeolusTech, 2021-10-04:
OK. So the problem is solved! It turns out that my firmware for Integrated Intel Graphics Card i915 was missing - that was the root cause of the problem. I used the _displaylink-debian.sh_ script to install displaylink again, installation froze, I've replace the .conf file with what @calebanthony suggested (Thanks a lot!) and voilla! It works. More detailed instructions below:

1. I started to mess with kernel versions
`sudo add-apt-repository ppa:cappelikan/ppa
sudo apt update
sudo apt install mainline`
I ran into some errors during kernel upgrade (missing dependencies etc.). But I have noticed that there was missing software for i915 - which led me to the guide mentioned in step 3.

2. I have uninstalled kernel 5.14 and 5.13 since they failed. However, the kernel failed upgrades broke my dpkg so I had to
`sudo dpkg --configure -a`

3. I started to install graphics driver following that [guide](https://itectec.com/ubuntu/ubuntu-how-to-get-and-install-intel-i915-drivers-on-ubuntu-18-04/).
`sudo add-apt-repository ppa:oibaf/graphics-drivers
 sudo apt update
sudo apt dist-upgrade`

4. Reboot
5. `sudo apt update && sudo apt -y upgrade`
6.  sudo ./displaylink-debian.sh --install
the installation froze though at the same step as earlier (_Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service -> /lib/systemd/system/displaylink-driver.service._)
7. switch to tty6 (Ctrl + Alt + F6)
8. `sudo reboot now`
9. the machine was literally dying after reboot, but I disconnected the HDMI and managed to reboot it one more time and it was a bit better.
10. `sudo ./displaylink-debian.sh --debug`

11.`sudo vim /etc/X11/xorg.conf.d/20-displaylink.conf`
`Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection`
12. `sudo reboot now`
13. works!

AeolusTech, 2021-10-04:
That's how the debug information looks when it is properly working.

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.11.0-37-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x13d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x11b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

closed 2021-10-04 by AeolusTech