AdnanHodzic/displaylink-debian #625: DisplayLink does not work on Ubuntu 20.4

MrDockal, 2021-09-27:
The installation fails on Ubuntu 20.4. dkms 2.8.1-5ubuntu2

#### 1)  When I run `sudo ./displaylink-debian.sh --install` for the first time I get the error
```
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-36-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4036?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2021-09-27 16:47:04--  https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 52.39.229.228, 34.210.13.121
Connecting to www.synaptics.com (www.synaptics.com)|52.39.229.228|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14500327 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.1.zip’

DisplayLink_Ubuntu_5.4.1.zip                  100%[=================================================================================================>]  13.83M   279KB/s    in 78s     

2021-09-27 16:48:23 (182 KB/s) - ‘DisplayLink_Ubuntu_5.4.1.zip’ saved [14500327/14500327]

Removing prior: "5.4.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.1.zip
  inflating: 5.4.1/displaylink-driver-5.4.1-55.174.run  
Creating directory displaylink-driver-5.4.1-55.174
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.1-55.174  100%  
patching file 5.4.1/displaylink-driver-5.4.1/displaylink-installer.sh
Hunk #1 succeeded at 456 (offset -1 lines).

Installing driver version: 5.4.1

DisplayLink Linux Software 5.4.1-55.174 install script called: install
Distribution discovered: Ubuntu 20.04.3 LTS

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.9.1/source ->
                 /usr/src/evdi-1.9.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.11.0-36-generic all INCLUDEDIR=/lib/modules/5.11.0-36-generic/build/include KVERSION=5.11.0-36-generic DKMS_BUILD=1...(bad exit status: 2)
ERROR (dkms apport): binary package for evdi: 1.9.1 not found
Error! Bad return status for module build on kernel: 5.11.0-36-generic (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.
ERROR: Failed to install evdi to the kernel tree.
```

#### 2) When I run the same script again I get a different error: 
```

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-36-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4036?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2021-09-27 16:50:33--  https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 52.39.229.228, 34.210.13.121
Connecting to www.synaptics.com (www.synaptics.com)|52.39.229.228|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14500327 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.1.zip’

DisplayLink_Ubuntu_5.4.1.zip                  100%[=================================================================================================>]  13.83M  1.74MB/s    in 9.2s    

2021-09-27 16:50:43 (1.50 MB/s) - ‘DisplayLink_Ubuntu_5.4.1.zip’ saved [14500327/14500327]

Removing prior: "5.4.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.1.zip
  inflating: 5.4.1/displaylink-driver-5.4.1-55.174.run  
Creating directory displaylink-driver-5.4.1-55.174
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.1-55.174  100%  
patching file 5.4.1/displaylink-driver-5.4.1/displaylink-installer.sh
Hunk #1 succeeded at 456 (offset -1 lines).

Installing driver version: 5.4.1

DisplayLink Linux Software 5.4.1-55.174 install script called: install
Distribution discovered: Ubuntu 20.04.3 LTS

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]
Error! DKMS tree already contains: evdi-1.9.1
You cannot add the same module/version combo more than once.
EVDI DKMS module already installed.
[[ Installing module configuration files ]]
[[ Installing EVDI library ]]
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.9.1 -lc -lgcc 
cp libevdi.so.1.9.1 libevdi.so
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

Installation complete!
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for details.
```

#### 3) And finally this is the debug output. I have noticed that there are 3 different versions under `Driver version:`

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.11.0-36-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.9.1
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [103c:880d]
VGA: Intel Corporation Device 9a49 (rev 01)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Initially I was following the official forum article describing how to install the driver (https://support.displaylink.com/knowledgebase/articles/1944022-how-to-install-displaylink-software-on-ubuntu-20-0) but unfortunately I got the same error as per step 1. That's why I guess there isn't an issue with your package, but rather my local environment. Any help would be appreciated. Thanks heaps!

Barabazs, 2021-09-30:
Hi @MrDockal! I would first try do remove all evdi versions before trying again.

Barabazs, 2021-10-19:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2021-10-19 by Barabazs

MrDockal, 2021-11-09:
Hello @Barabazs thank you for your reply .. Sorry I was out of the office for a while where the docking station is.
I tried to uninstall all evdi versions, but the same bug persisted. 

```
Creating symlink /var/lib/dkms/evdi/1.9.1/source ->
                 /usr/src/evdi-1.9.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.11.0-36-generic all INCLUDEDIR=/lib/modules/5.11.0-36-generic/build/include KVERSION=5.11.0-36-generic DKMS_BUILD=1...(bad exit status: 2)
ERROR (dkms apport): binary package for evdi: 1.9.1 not found
Error! Bad return status for module build on kernel: 5.11.0-36-generic (x86_64)
Consult /var/lib/dkms/evdi/1.9.1/build/make.log for more information.
```

However, I ended up upgrading my kernel to 5.11.0-40-generic and the installation seems to be successful this time. As soon as I try it with the docking station I'll let you know.