AdnanHodzic/displaylink-debian #627: Crash in dma-buf driver on kernel 5.14+nvidia when resizing/repositioning

ApeironTsuka, 2021-10-01:
I'm not sure if this is something I should be reporting here or over in the upstream evdi github.

The issue: Under kernel 5.10, everything works fine. Everything else equal, it causes the below fault under 5.14 when resizing, repositioning, and sometimes when enabling the screen in the first place (--setprovideroutputsource calls).

```
kernel: ------------[ cut here ]------------
kernel: kernel BUG at drivers/dma-buf/dma-buf.c:1336!
kernel: invalid opcode: 0000 [#1] SMP PTI
kernel: CPU: 6 PID: 1077 Comm: Xorg Tainted: P          IOE     5.14.0-1-amd64 #1  Debian 5.14.6-2
kernel: Hardware name: LENOVO 20217/20217, BIOS 74CN35WW(V1.10) 07/18/2013
kernel: RIP: 0010:dma_buf_vunmap+0x7a/0x90
kernel: Code: 0b 48 89 ee 48 89 df e8 d4 4f 59 00 80 7b 58 00 48 c7 43 50 00 00 00 00 74 04 c6 43 58 00 5b 4c 89 e7 5d 41 5c e9 d6 db 29 00 <0f> 0b 0f 0b c3 0f 0b >
kernel: RSP: 0018:ffffbc1580f2bb40 EFLAGS: 00010202
kernel: RAX: ffffbc1585001000 RBX: ffff917d49bcb800 RCX: 0000000000000000
kernel: RDX: 0000000000000000 RSI: ffffbc1580f2bb60 RDI: ffff917d49bcb800
kernel: RBP: ffffbc1580f2bb60 R08: ffff917d433e6e18 R09: ffff917d40e901f8
kernel: R10: 0000000000000000 R11: ffff917d41a83d98 R12: ffff917d4984ea00
kernel: R13: ffff917d58c24c00 R14: ffff917d58c24b40 R15: 0000000000000000
kernel: FS:  00007f85aedcba40(0000) GS:ffff917ea6f80000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 00007f859f02c000 CR3: 00000001197a8006 CR4: 00000000001706e0
kernel: Call Trace:
kernel:  evdi_gem_vunmap+0xae/0xc0 [evdi]
kernel:  ? drm_mode_object_register+0x10/0x40 [drm]
kernel:  evdi_gem_free_object+0x18/0xb0 [evdi]
kernel:  evdi_user_framebuffer_destroy+0x4a/0x60 [evdi]
kernel:  evdi_plane_atomic_update+0xd7/0x270 [evdi]
kernel:  ? __cond_resched+0x16/0x40
kernel:  drm_atomic_helper_commit_planes+0xc2/0x250 [drm_kms_helper]
kernel:  drm_atomic_helper_commit_tail+0x42/0x80 [drm_kms_helper]
kernel:  commit_tail+0xce/0x130 [drm_kms_helper]
kernel:  drm_atomic_helper_commit+0x113/0x140 [drm_kms_helper]
kernel:  drm_atomic_helper_set_config+0x70/0xb0 [drm_kms_helper]
kernel:  drm_mode_setcrtc+0x1d3/0x6d0 [drm]
kernel:  ? drm_mode_getcrtc+0x180/0x180 [drm]
kernel:  drm_ioctl_kernel+0xaa/0xf0 [drm]
kernel:  drm_ioctl+0x220/0x3c0 [drm]
kernel:  ? drm_mode_getcrtc+0x180/0x180 [drm]
kernel:  __x64_sys_ioctl+0x83/0xb0
kernel:  do_syscall_64+0x3b/0xc0
kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xae
kernel: RIP: 0033:0x7f85af237957
kernel: Code: 3c 1c 48 f7 d8 4c 39 e0 77 b9 e8 24 ff ff ff 85 c0 78 be 4c 89 e0 5b 5d 41 5c c3 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 >
kernel: RSP: 002b:00007ffecf16f688 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
kernel: RAX: ffffffffffffffda RBX: 00007ffecf16f6c0 RCX: 00007f85af237957
kernel: RDX: 00007ffecf16f6c0 RSI: 00000000c06864a2 RDI: 0000000000000015
kernel: RBP: 00000000c06864a2 R08: 0000000000000000 R09: 00005578eb04df20
kernel: R10: 0000000000000000 R11: 0000000000000246 R12: 00005578ea703ef0
kernel: R13: 0000000000000015 R14: 00005578ea6f5a68 R15: 0000000000000001
kernel: Modules linked in: vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg intel_rapl_msr intel_rapl_comm>
kernel:  mei_me watchdog mei ideapad_laptop nvidia_drm(POE) platform_profile sparse_keymap rfkill ac evdev nvidia_modeset(POE) nvidia(POE) ipmi_devintf ipmi_msghan>
kernel: ---[ end trace 06292ed45b30dd9e ]---
```
Here's the debug output for 5.14: (it's identical to 5.10 aside from kernel saying 5.10.0-8-amd64)
I don't know offhand why it's saying the release is sid. It's Bookworm, the sid repos aren't in use.
```
# ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: sid
Kernel: 5.14.0-1-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: GT
VGA: NVIDIA Corporation GK107M [GeForce GT 750M] (rev a1)
VGA (3D): NVIDIA Corporation GK107M [GeForce GT 750M] (rev a1)
X11 version: 1.20.11-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x279 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0x32a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x308 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2e6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2c4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Barabazs, 2021-10-02:
Hi @ApeironTsuka you should report this upstream. There is not much we can do here.

closed 2021-10-02 by Barabazs

ApeironTsuka, 2021-10-02:
Ahh, figured that might be the case. Thanks, I'll head over there.