AdnanHodzic/displaylink-debian #631: X11 not showing additional screens after apt upgrade

andrewroffey, 2021-10-15:
Hi,

I have a Debian Bullseye system which previously had displaylink-debian working prior to running `apt upgrade` today (great project by the way).

After patching my system today I'm unable to see the additional screens via `arandr` or the KDE Display Configuration. `arandr` only shows my laptop screen eDP-1.

FWIW, when I shut down my system I can see output across all of my monitors via Plymouth.

I've tried uninstalling and re-installing displaylink-debian with the latest version from Git, no change (I did uninstall, reboot, install, reboot).

I've tried [this](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348) and [this](https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411) and rebooted each time, no change.

I've tried running these to attach the providers to the main provider:

```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0
```

Here's my debug output:

```
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-9-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation TigerLake GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 1.20.11-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x4c cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 9 associated providers: 4 name:modesetting
Provider 1: id: 0x152 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x130 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x10e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xec cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

Any assistance greatly appreciated... thanks!

AnthonyAportela, 2021-12-13:
Having almost exactly the same problem (I think) with almost the exact same debug output:
`--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.11.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9bc8 (rev 03)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x10e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xec cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xca cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xa8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
`

rotten, 2021-12-20:
I tried a variety of permutation on  `20-displaylink.conf` to no avail.   Currently I'm on the "PageFlip" variation.  I tried reinstall, uninstall, and multiple reboots.

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-91-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nouveau
Subsystem: UHD
GP106M
VGA: Intel Corporation UHD Graphics 630 (Mobile)
NVIDIA Corporation GP106M [GeForce GTX 1060 Mobile] (rev a1)
VGA (3D):
X11 version: 1.20.13-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x43 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 1 associated providers: 5 name:modesetting
Provider 1: id: 0x15c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xf6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xd4 cap: 0x6, Sink Output, Source Offload crtcs: 4 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Fun side note:  `vim` highlights `ENDSECTION` in bright red by default as if it were an error.  If you put a space between "END" and "SECTION", vim is happy, but then X won't start.

rotten, 2021-12-28:
FWIW, I uninstalled (and rebooted), then installed the new Beta release from December 22nd using the `*.run` script from the Display link site (and rebooted).  That left me with no 20-displaylink.conf file.  It didn't work, so I dropped the 20-displaylink.conf file in as shown above.  It still didn't work (after reboot).  Whenever I try to use the script in this repo it complains about the beta version and tries to revert to an older version.  The displaylink-debian.sh script does not have an option to force Beta without editing the script.

It was worth a try.  I'm game to dig into or poke around or answer questions if you can give me some advice as to where else to look or what else to experiment with.

rotten, 2022-01-07:
I saw there was a patch submitted to ubuntu over the holidays that addressed some sort of issue with DisplayLink drivers.  The updated patches were released this week.  This afternoon I applied them.  I re-installed the drivers, updated the config with that PageFlip setting and now it is working !!  happy dance.

Barabazs, 2022-01-30:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2022-01-30 by Barabazs

andrewroffey, 2022-03-18:
Thanks for your suggestions and apologies for the very late reply.

I've updated to the latest version, 1.10.1, and tried the three combinations (no 20-displaylink.conf file, [this](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348) and [this](https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411)) and same result - no monitors available in xrandr/arandr. 

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-12-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.1
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation TigerLake GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x4c cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 9 associated providers: 4 name:modesetting
Provider 1: id: 0x158 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x136 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x114 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xf2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

rotten, 2022-03-18:
Another thing I noticed was that I had an older display link adapter (8 years old?) and a new one (that I just bought).  The older one didn't work, but the new one did.  I thought maybe I had a bad adapter - I tried it on my wife's windows system, and while the software recognized it, it didn't work there either.

Now I'm wondering if there is something in the new drivers that breaks compatibilities with the old adapters.

This cheap knockoff worked (on both my linux system and my wife's windows system):  https://amazon.com/gp/product/B08XQGFP3N/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1

andrewroffey, 2022-05-06:
Eventually figured out my problem.

I had a monitor plugged directly into my laptop via HDMI. Once I disconnected that I was able to connect my monitors via DisplayLink.

(I was only able to plug in one monitor directly via HDMI, so I'm now back to 2x monitors + laptop screen)