AdnanHodzic/displaylink-debian #643: ZorinOS Pro 16 Driver Request

lavabrothers, 2021-11-01:
Here is the output of the information 'sudo ./displaylink-debian.sh --debug' as requested when providing a support request

--------------- Linux system info ----------------

Distro: Zorin
Release: focal
Kernel: 5.11.0-38-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Ltd
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] (rev c1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x59 cap: 0x9, Source Output, Sink Offload crtcs: 6 outputs: 4 associated providers: 0 name:AMD Radeon RX 5700 XT @ pci:0000:0b:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
Hello! I was wondering if it was possible to get the debian displaylink driver to work on Zorin? It works wonderfully for my Linux Mint 20 install and you do amazing work!
If that is not possible, all good. Thank you for reading and considering!

Barabazs, 2021-11-02:
Hi @lavabrothers I've updated the script. Can you try again please?

closed 2021-11-02 by Barabazs

lavabrothers, 2021-11-02:
It installed well, same as Linux Mint. For some reason I was unable to get it working unfortunately. There is no display out of the d6000 dock

PR #644 (2021-11-02): Add support for ZorinOS
Closes #643