AdnanHodzic/displaylink-debian #645: AMD GPU, DisplayLink Device not working

mh254, 2021-11-08:
Here is the Debug Output:
--------------- Linux system info ----------------

Distro: elementary
Release: hera
Kernel: 5.4.0-89-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.9.1
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: R6/R7
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Kaveri [Radeon R6/R7 Graphics]
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2~18.04.5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x123 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x101 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xdf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xbd cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

No errors while installing the DisplayLink software/driver but the display is not working.

Barabazs, 2021-12-21:
Can you provide a more detailed description please.
The debug output does not show anything abnormal.

mh254, 2021-12-21:
What information do you need?

Driver installation was without any error.

When i connect the USB Adapter, the attached screen doesn't work.

On Windows it is working.

closed 2022-02-03 by Barabazs