AdnanHodzic/displaylink-debian #647: System freezes when using NVIDIA drivers | Elementary Odin

isc-joserodriguez, 2021-11-15:
My problem is that when I have the NVIDIA drivers installed, when doing a screen configuration the system freezes.

The same happens if I have the displaylink device connected during system boot, I must connect it until I enter the system for it to work.

I need to put a monitor upright and I can't because the system freezes when trying to change the monitor settings.

Tried with nvidia-driver-495 and nvidia-driver-470 with no success.

My system info:
``` bash
GF63-Thin-10SC-652d3676
Intel® Core™ i5-10200H CPU @ 2.40GHz
NVIDIA GeForce GTX 1650 with Max-Q Design/PCIe/SSE2
```

Here is the debug output:

``` bash
--------------- Linux system info ----------------

Distro: Elementary
Release: odin
Kernel: 5.11.0-40-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation Device 9ba4 (rev 05)
VGA (3D): NVIDIA Corporation Device 1f9d (rev a1)
X11 version: 1.20.11-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2c1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x29f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x27d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x25b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1dd cap: 0x2, Sink Output crtcs: 3 outputs: 2 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

YushchenkoAndrew, 2021-12-08:
Had the same issue as you with NVIDIA gpu. Switching display protocol from X11 to Wayland resolved it. Hope it'll help you.

isc-joserodriguez, 2021-12-09:
@YushchenkoAndrew Thanks! 

Elementary does not support wayland. I'm going to install another distro and try that (probably Pop Os) What distro do you use?

YushchenkoAndrew, 2021-12-09:
@isc-joserodriguez  Im actually using Pop os and so far havnt seen some major issues

To enable wayland in Pop os its quite simple, just comment the line "WaylandEnable=false" in the file /etc/gdm3/custom.conf and relogin

But you should be aware that it may be not so stable with some apps. (Some game just wont run with wine)

ApeironTsuka, 2021-12-28:
This was fixed in the upstream 1.10.0 devel version of the evdi kernel module (as well as the 5.5 beta driver version). See my (closed) PR over there for the fix if you want to manually apply it to an existing 1.9.1 (it's not a big patch).

closed 2022-01-30 by Barabazs