AdnanHodzic/displaylink-debian #651: /etc/X11/xorg.conf.d/20-displaylink.conf breaks external displays

jtmackoy, 2021-12-01:
Ran into a problem whereby my Plugable ud-ultc-4k dock stopped displaying two external monitors after an upgrade to 5.15.5-76051505-generic of the pop-os kernel.  Ultimately, the problem was due to evdi not compiling properly when dkms tried to update modules for the new kernel.  [This proved a sufficient workaround for now](https://github.com/DisplayLink/evdi/issues/308#issuecomment-981731578) until the [evdi devs release an updated version sometime in December](https://github.com/DisplayLink/evdi/issues/308#issuecomment-979803934).

Opened this issue because /etc/X11/xorg.conf.d/20-displaylink.conf created by displaylink-debian had to be removed in order for the workaround to be complete.

Here are the debug details on my working system:
```
--------------- Linux system info ----------------

Distro: Pop
Release: hirsute
Kernel: 5.15.5-76051505-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1558:50e1]
[1558:50e1]
VGA: Intel Corporation Device 9a60 (rev 01)
NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 5 name:modesetting
Provider 1: id: 0x313 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2f1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2c4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x29c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x272 cap: 0x2, Sink Output crtcs: 4 outputs: 4 associated providers: 1 name:NVIDIA-G0

-------------------------------------------------------------------
```

greatlord, 2021-12-30:
I think the beta 5.5 have solv that issue I just got it compile for debian after some changes in the installar and so see my open issue for debian I just posted.  I did manual sign the module as well. I post my sign script as well

Barabazs, 2022-01-30:
Kernel 5.15 is not supported by DisplayLink stable version.
I'm not planning on supporting a beta release.

closed 2022-01-30 by Barabazs