AdnanHodzic/displaylink-debian #652: External displays become very slow when laptop is closed

aa777888-2, 2021-12-02:
First of all, and most important, many thanks for your hard work on this! 

Overall config: Dell XPS 13 laptop running Kali, Dell Displaylink USB 3.0 dock, two Dell external monitors. Extended desktop, left to right: Ext Mon 1 -- Ext Mon 2 -- Laptop.

Note: I had to delete /etc/X11/xorg.conf.d/20-displaylink.conf to get everything to work--I saw that in another open ticket, and you'll also see the relevant warning in the debug output below.

**Behavior: when the laptop lid is closed all screen drawing on the external monitors becomes very slow.**

Things I tried:

- Removed laptop display from display setup in the Kali Xfce Display Setup app--no change (in fact I had to put it back into the mix to get things to run smooth).
- Set all /etc/systemd/logind.conf HandleLidSwitch entries to ignore.
- Looked at Xfce settings, all are set to ignore lid state.
- Checked XPS 13 BIOS to see if there was a way to defeat lid switch (nope!).

Relevant lsusb output:

Bus 003 Device 003: ID 17e9:4318 DisplayLink Dell USB 3.0 Dock

Operating system:

$ lsb_release -a
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2021.4
Codename:       kali-rolling

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.14.0-kali4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.11-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0x139 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x117 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

$ xrandr --listproviders                   
Providers: number : 5
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0x139 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x117 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

And this is the script I run on login to make it all automated:

xrandr --setprovideroutputsource 1 0 && xrandr --setprovideroutputsource 2 0 && xrandr --setprovideroutputsource 3 0 && xrandr --setprovideroutputsource 4 0

rfb, 2021-12-05:
Hi, I'm experiencing the same thing. Display is fine when the lid is open, but very slow when the lid is closed. XPS P54G.

```
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-9-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.0+dfsg
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.11-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x14a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x128 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x106 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

rfb, 2021-12-05:
Hi @aa777888-2 - [Found this comment in prior issues.](https://github.com/AdnanHodzic/displaylink-debian/issues/393#issuecomment-906745453) It fixed the problem for me!

aa777888-2, 2021-12-05:
> Hi @aa777888-2 - [Found this comment in prior issues.](https://github.com/AdnanHodzic/displaylink-debian/issues/393#issuecomment-906745453) It fixed the problem for me!

Ryan--sadly this did not work for me. Still the same behavior with or without adding adding LIBGL_DRI3_DISABLE=true
to /etc/environment :-( :-(

rfb, 2021-12-05:
Shooting in the dark here, but if it is a DRI3 issue maybe this would help: https://wiki.archlinux.org/title/intel_graphics#DRI3_issues

aa777888-2, 2021-12-05:
Thanks, but it's not helping. It does not appear to be a DRI3 issue.

t-walter, 2022-01-01:
> Hi @aa777888-2 - [Found this comment in prior issues.](https://github.com/AdnanHodzic/displaylink-debian/issues/393#issuecomment-906745453) It fixed the problem for me!

Yay, finally the solution that worked for me, thanks a lot @rfb!
But my hardware differs from yours @aa777888-2. I have Lenovo Yoga Slim7 with AMD graphics running Mint. Finally, compared to standard configuration, I only had to: disable secure boot, install displaylink-debian, remove /etc/X11/xorg.conf.d/20-displaylink.conf and add LIBGL_DRI3_DISABLE=true to /etc/environment. Hope you will find the solution one day.

Barabazs, 2022-01-30:
Closing due to out of scope.

closed 2022-01-30 by Barabazs