AdnanHodzic/displaylink-debian #654: Only One Provider listed in Debian "testing aka bookworm" => evdi service version not found

loiselqw, 2021-12-10:
Dear community, 

I have here two machine both thinkpads with onboard intel graphics chip. After getting latest kernel update displaylink is not working. In both cases it looks like displaylink service is not running properly - do you have any suggestions in fixing this issue?

debug outputs are attached for both machines:

`--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 5.15.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.0.0
1.9.1
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.13-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting`

and

`--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 5.15.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.13-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver	"modesetting"
    Option	"AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------`

I tried various /etc/X11/xorg.conf.d/20-displaylink.conf with default "intel" "intel" setup and others described in common issues, but same behaviour as described.

Thx a lot for your suggestions

christian-krieg, 2021-12-10:
Hi,

I am having the same issue. To solve the `evdi` issue, install the EVDI kernel module:

```
sudo apt install evdi-dkms
```

You can load the module with:
```
sudo modprobe evdi
```
You should now be able to see EVDI activated in `displylink-debian`'s debug output.

To permanently load the module on each startup, add it to `/etc/modules`:
```
sudo su -c "echo evdi >> /etc/modules"
```

christian-krieg, 2021-12-10:
Hey there, I upgraded from Debian "buster" to Debian "bullseye". I first just re-installed `displaylink-debian` after pulling, but it turned out this was not enough. What worked for me was to uninstall `displaylink-debian`, reboot, and install again. This forces the tool to download and build from the latest driver sources. After reboot, `xrandr --listproviders` shows my screens again and everything works as before the upgrade.

Hope that helps!

loiselqw, 2021-12-11:
Unfortunately it works again, I had to do following to get it work:

- uninstall displaylink-debian 
-  reboot
- Clone the fork of evdi from nilathedragon:
`git clone https://github.com/nilathedragon/evdi`
- Copy the contents of evdi/modules to /var/lib/dkms/evdi/1.9.1/source/
`sudo cp -r ./evdi/module/* /var/lib/dkms/evdi/1.9.1/source/`
- install displaylink-debian 

everything descibed here:

https://github.com/DisplayLink/evdi/issues/308

closed 2021-12-11 by loiselqw