AdnanHodzic/displaylink-debian #657: DisplayLink service status blank and no EVDI Version found

thov2021, 2021-12-14:
Hallo everyone,

First of all, thank you in advance for your help. I've tried reinstalling the package and searching previous tickets, why there is no EVDI service version being found but couldn't find anything useful.

I got an error once while re-installing that the EVDI version couldn't be installed. The error message was "Failed to installed EVDI to kernel tree". I haven't been able to reproduce it since.

Attached is the debug results I got from running this command: "sudo ./displaylink-debian.sh --debug". This was run with the USB connected to the Dock Pro 3.0.

Thanks in advance for your help.

[Debug Output.docx](https://github.com/AdnanHodzic/displaylink-debian/files/7709716/Debug.Output.docx)
.

Barabazs, 2021-12-15:
Please copy-paste the debug output in this issue.

thov2021, 2021-12-15:
> Please copy-paste the debug output in this issue.

Thank you for your reply. Please find the output attached. Sorry I didn't know how to do this when I created this Issue
[Debug Output.zip](https://github.com/AdnanHodzic/displaylink-debian/files/7718248/Debug.Output.zip)
.

chrros95, 2021-12-15:
The current release of evdi (1.9.1) doesn't support the kernel 5.15 because a dependency was changed.
You can try to the latest commit that already contains a fix by following these steps:

```
$PATH/displaylink-debian.sh --uninstall
cd /usr/src/
rm -rf evdi*
git clone --depth=1 https://github.com/DisplayLink/evdi.git
ln -s evdi/module evdi-1.9.1
dkms build evdi/1.9.1
$PATH/displaylink-debian.sh --install
```
`PATH` should be the directory where you saved this repo.

With this my screen are back online with the same OS an kernel but the screens are now lagging and I don't have a solution for that yet.

thov2021, 2021-12-20:
> The current release of evdi (1.9.1) doesn't support the kernel 5.15 because a dependency was changed. You can try to the latest commit that already contains a fix by following these steps:
>
> ``` [...]

thov2021, 2021-12-20:
Accidentally pressed the close button!

Barabazs, 2021-12-21:
Please create a new issue with the correct debug output. (not as an attachment)

closed 2021-12-21 by Barabazs

matthewhegarty, 2022-01-24:
> With this my screen are back online with the same OS an kernel but the screens are now lagging and I don't have a solution for that yet.

I have the same issue.  My 3rd monitor wasn't working but [this solution](https://github.com/AdnanHodzic/displaylink-debian/issues/657#issuecomment-997789579) fixed it.  Now there is minor lag on the 3rd monitor.