AdnanHodzic/displaylink-debian #658: Mouse and keyboard lag/slow with Desktop PC, primary and secondary monitor connected

codinesh, 2021-12-17:
I have a Desktop PC (AMD and Nvidia graphics). After following the installation steps, both the monitors are connected, but have lagging/performance issue with mouse and keyboard input.

I see there are similar issues raised for laptop, when lid is closed (primary display is turned off), in this case, its a desktop pc and primary and secondary monitors are connected.

<pre>--------------- Linux system info ----------------

Distro: Ubuntu
Release: impish
Kernel: 5.13.0-22-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: Ltd
VGA: NVIDIA Corporation GK208B [GeForce GT 710] (rev a1)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section &quot;ServerLayout&quot;
    Identifier &quot;layout&quot;
    Screen 0 &quot;nvidia&quot;
    Inactive &quot;intel&quot;
EndSection

Section &quot;Device&quot;
    Identifier &quot;intel&quot;
    Driver &quot;modesetting&quot;
    Option &quot;AccelMethod&quot; &quot;None&quot;
EndSection

Section &quot;Screen&quot;
    Identifier &quot;intel&quot;
    Device &quot;intel&quot;
EndSection

Section &quot;Device&quot;
    Identifier &quot;nvidia&quot;
    Driver &quot;nvidia&quot;
    Option &quot;ConstrainCursor&quot; &quot;off&quot;
EndSection

Section &quot;Screen&quot;
    Identifier &quot;nvidia&quot;
    Device &quot;nvidia&quot;
    Option &quot;AllowEmptyInitialConfiguration&quot; &quot;on&quot;
    Option &quot;IgnoreDisplayDevices&quot; &quot;CRT&quot;
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x278 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0x320 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2fe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2cf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
</pre>

evscoding, 2022-01-26:
I have the same on my laptop. Very high cpu usage.

Barabazs, 2022-01-30:
You'll have to report this upstream. Not much we can do about it here.

closed 2022-01-30 by Barabazs