AdnanHodzic/displaylink-debian #661: DisplayLink service status blank and no EVDI Version found

thov2021, 2021-12-22:
Hello everyone,

First of all, thank you in advance for your help. I've tried reinstalling the package and searching previous tickets, why there is no EVDI service version being found but couldn't find anything useful.

I got an error once while re-installing that the EVDI version couldn't be installed. The error message was "Failed to installed EVDI to kernel tree". I haven't been able to reproduce it since.

Attached is the debug results I got from running this command: "sudo ./displaylink-debian.sh --debug". This was run with the USB connected to the Dock Pro 3.0.

Thanks in advance for your help.

```--------------- Linux system info ----------------

Distro: Pop
Release: focal
Kernel: 5.15.5-76051505-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: 620
Integrated
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting```

Barabazs, 2021-12-27:
What is the output of `dkms status`

thov2021, 2021-12-27:
> What is the output of `dkms status`

The following is the output of `dkms status`
```
evdi, 1.9.1: added
system76, 1.0.13~1630342318~20.04~68bd479, 5.11.0-7620-generic, x86_64: installed
system76, 1.0.13~1630342318~20.04~68bd479, 5.11.0-7633-generic, x86_64: installed
system76, 1.0.13~1630342318~20.04~68bd479, 5.13.0-7614-generic, x86_64: installed
system76, 1.0.13~1630342318~20.04~68bd479, 5.13.0-7620-generic, x86_64: installed
system76, 1.0.13~1630342318~20.04~68bd479, 5.15.5-76051505-generic, x86_64: installed
system76_acpi, 1.0.2~1635519829~20.04~b49caa5, 5.13.0-7614-generic, x86_64: installed
system76_acpi, 1.0.2~1635519829~20.04~b49caa5, 5.13.0-7620-generic, x86_64: installed
system76_acpi, 1.0.2~1635519829~20.04~b49caa5, 5.15.5-76051505-generic, x86_64: installed
system76-io, 1.0.1~1616424744~20.04~57c15ff, 5.11.0-7620-generic, x86_64: installed
system76-io, 1.0.1~1616424744~20.04~57c15ff, 5.11.0-7633-generic, x86_64: installed
system76-io, 1.0.1~1616424744~20.04~57c15ff, 5.13.0-7614-generic, x86_64: installed
system76-io, 1.0.1~1616424744~20.04~57c15ff, 5.13.0-7620-generic, x86_64: installed
system76-io, 1.0.1~1616424744~20.04~57c15ff, 5.15.5-76051505-generic, x86_64: installed
wireguard, 1.0.20201112: added
´´´

ApeironTsuka, 2021-12-28:
Version 1.9.1 of the evdi module doesn't build under kernel 5.15. You'll need to either use the 5.5 beta version of the driver, or clone the upstream devel repo for evdi for fixes. Or check the PRs there, as there are a couple for 5.15 support you could manually make the changes from in /usr/src/evdi-1.9.1.

greatlord, 2021-12-30:
I just modify so this script can build and install the beta version 5.5 so it works on kernel 5.15
I you can unpack it and build it for kernel 5.15 without any issue. 
it contain one hack and one revert and update patch for 5.5 evid installer as well. 
this can not build older version only 5.5 beta and maybe newer as well
[displaylink-debian.zip](https://github.com/AdnanHodzic/displaylink-debian/files/7791091/displaylink-debian.zip)

thov2021, 2022-01-06:
> Version 1.9.1 of the evdi module doesn't build under kernel 5.15. You'll need to either use the 5.5 beta version of the driver, or clone the upstream devel repo for evdi for fixes. Or check the PRs [...]

Thank you very much for your message. I'm not really an advanced user of Ubuntu/ Linux so I don't know what "PRs" mean. Could you explain that please? I think I'd like to try out your last suggestion first before using the beta driver version.

Thank you in advance!

closed 2022-01-06 by thov2021