AdnanHodzic/displaylink-debian #662: New DisplayLink Release [5.5-beta] breaks strategy displaylink-debian.sh script

bopnrgi, 2021-12-22:
A new update comes out today and breaks script.
Filename doesn't match regex pattern, because of "beta" in filename

`Downloading DisplayLink Ubuntu driver:

--2021-12-22 20:05:57--  https://www.synaptics.com/sites/default/files/exe_files/2021-12/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%20%28Beta%295.5%20Beta-EXE.zip
Résolution de www.synaptics.com (www.synaptics.com)... 52.41.97.53, 34.214.248.103
Connexion vers www.synaptics.com (www.synaptics.com)|52.41.97.53|:443... connecté.
requête HTTP transmise, en attente de la réponse... 200 OK
Taille : 16894726 (16M) [application/zip]
Enregistre : «DisplayLink_Ubuntu_5.5.zip»

DisplayLink_Ubuntu_5.5.zip            100%[=======================================================================>]  16,11M  1,14MB/s    ds 27s     

2021-12-22 20:06:25 (609 KB/s) - «DisplayLink_Ubuntu_5.5.zip» enregistré [16894726/16894726]

Removing prior: "5.5" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.5.zip
  inflating: 5.5/DisplayLink USB Graphics Software for Ubuntu 5.5.0-beta-59.118-Release Notes.txt  
  inflating: 5.5/displaylink-driver-5.5.0-beta-59.118.run  
chmod: impossible d'accéder à '5.5/displaylink-driver-5.5-[0-9]*.run': Aucun fichier ou dossier de ce type
`

the-jaxon, 2021-12-23:
Confirming same issue.  Appears to be around line 436 in displayport-debian.sh

mh254, 2021-12-27:
Confirmed, same issue here....

jeffgirard, 2021-12-27:
Confirmed, same issue here too, attempted to fix this on my side (failed!), thanks in advance to the person who can fix that so we can benefit from it quickly :)

Barabazs, 2021-12-27:
I updated the script. Please retry with the latest version from the repo.

jeffgirard, 2021-12-27:
I confirm its now working, script running smoothly until completion, thanks a lot for that.

the-jaxon, 2021-12-29:
Second confirmation.  Debian 11, working great again!  Thank you!

closed 2022-01-06 by Barabazs