AdnanHodzic/displaylink-debian #663: Script cannot run in Debian 11.2

AngelBlue01, 2021-12-25:
I just upgraded my system from Debian 10 to 11. Of course I had to reinstall the DisplayLink drivers, which worked well in Debian 10. I downloaded the latest version of main from Github as a zip and extracted the contents. However, `sudo sh ./displaylink-debian.sh` will not run, I get the following errors:

> displaylink-debian.sh: 26: Bad substitution
> displaylink-debian.sh: 98: Syntax error: "(" unexpected (expecting "fi")

Barabazs, 2021-12-27:
Can you retry with the latest version from the repo please?

AngelBlue01, 2021-12-28:
With the latest main I now get this output:

> displaylink-debian.sh: 23: [[: not found
> displaylink-debian.sh: 34: Bad substitution
> displaylink-debian.sh: 106: Syntax error: "(" unexpected (expecting "fi")

Barabazs, 2022-01-06:
Can you just run this `sudo ./displaylink-debian.sh`