AdnanHodzic/displaylink-debian #664: I need the beta 5.5 driver for get it work with kernel 5.15 in debian bookworm / sid

greatlord, 2021-12-30:
Hi
Only the beta driver work and you need a new patch as well for the beta driver 
I attach my patch here for the beta driver 

[displaylink-installer.zip](https://github.com/AdnanHodzic/displaylink-debian/files/7791042/displaylink-installer.zip)

Next issue I did have was I did create a bit ugly hack in the debian installer script as well
file:///home/magnus/git/displaylink-debian/displaylink-debian.zip
remove the  beta driver detactions and modify follow line at 437  to 
chmod +x $driver_dir/displaylink-driver-*.run
./$driver_dir/displaylink-driver-*.run --keep --noexec
mv displaylink-driver-*/ $driver_dir/displaylink-driver-${version}

Next issue I got i need the driver be sign so i wrote debian sign script for the driver for kernel 5.15 
I should update so it work for all kernel 

see the comment in the script i took thuse comment from debian wiki how to sign you own module,
if you already create you own sign key already or you have install virtualbox do not create sign key and enroll it
only run the mdoule sign part.
[debian.zip](https://github.com/AdnanHodzic/displaylink-debian/files/7791062/debian.zip)

Maybe I will commit it to a fork all change I done so people can get it. 

please do not compain about the script it is very fast writen it should have more check to make it safe. 
so it does not overwrite virtualbox private sign key or other thrid party module sign.
you ahould only have one own private sign key for modules and kernel.

syswombat, 2022-01-04:
yes plaease!  ;-)

Barabazs, 2022-01-06:
Hi @greatlord. Feel free to open a PR with your fixes.

closed 2022-01-06 by Barabazs