AdnanHodzic/displaylink-debian #666: Problem with Kernel: 5.15.0-2-amd64 x86_64 using a IcyBox 14-in-1 USB Type-C | IB-DK2262AC

syswombat, 2022-01-04:
hello
your script did a error!
My Installed Debian is Testing because i need it because of the WiFi card build in

Kernel: 5.15.0-2-amd64 x86_64 bits: 64 Console: tty 1 Distro: Debian GNU/Linux 11 (bullseye) 
Machine:   Type: Laptop System: LENOVO product: 20YGCTO1WW v: ThinkPad E15 Gen 3 

Graphics:  Device-1: Advanced Micro Devices [AMD/ATI] Lucienne driver: amdgpu v: kernel 
           Device-2: IMC Networks Integrated Camera type: USB driver: uvcvideo 
           Display: server: X.Org 1.20.11 driver: loaded: amdgpu,ati unloaded: fbdev,modesetting,vesa resolution: 
           1: 1920x1080~60Hz 2: 1920x1200~60Hz 
           OpenGL: renderer: AMD RENOIR (DRM 3.42.0 5.15.0-2-amd64 LLVM 11.0.1) v: 4.6 Mesa 20.3.5

Using a Dockingstation
IcyBox 14-in-1 USB Type-C | IB-DK2262AC
           
----------------------
Checking dependencies

unzip is installed
dpkg-query: Paket »linux-headers-5.15.0-2-amd64« ist nicht installiert und es ist keine Information verfügbar
Verwenden Sie dpkg --info (= dpkg-deb --info) zum Untersuchen von Archiven.
linux-headers-5.15.0-2-amd64 not found! Install? [Y/n] y
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 gcc-11 : Hängt ab von: cpp-11 (= 11.2.0-13) soll aber nicht installiert werden
          Hängt ab von: libcc1-0 (>= 11.2.0-13) aber 10.2.1-6 soll installiert werden
          Hängt ab von: binutils (>= 2.37) aber 2.35.2-2 soll installiert werden
          Hängt ab von: libgcc-11-dev (= 11.2.0-13) soll aber nicht installiert werden
          Hängt ab von: libc6 (>= 2.33) aber 2.31-13+deb11u2 soll installiert werden
 linux-kbuild-5.15 : Hängt ab von: libc6 (>= 2.33) aber 2.31-13+deb11u2 soll installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
linux-headers-5.15.0-2-amd64 installation failed.  Aborting.

syswombat, 2022-01-06:
`
$ sudo ./displaylink-debian.sh --debug
Beta version detected, trying to download previous version

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.15.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Lucienne (rev c1)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Datei oder Verzeichnis nicht gefunden
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:05:00.0

`

syswombat, 2022-01-11:
any idea or help, or are more information needed?

syswombat, 2022-01-14:
do you need more information i could help?

rogue73, 2022-01-17:
Hi,

i've found this post: https://github.com/DisplayLink/evdi/issues/308#issuecomment-981731578
With this information I've got it to get Displaylink (i-tec) running with my installation (debian bookworm/sid) and  currently i'm using kernel Linux  5.15.0-2-amd64 #1 SMP Debian 5.15.5-2 (2021-12-18) x86_64 GNU/Linux

Barabazs, 2022-01-30:
Kernel 5.15 is not supported by DisplayLink stable version.
I'm not planning on supporting a beta release.

closed 2022-01-30 by Barabazs