AdnanHodzic/displaylink-debian #669: Unsatisfied dependencies. Missing component: DKMS.

nullquine, 2022-01-06:
Hello Adnan,

I encountered the following proble: when trying to run the displaylink-debian.sh script (as root) it reaches to the point of Preparing for install, but after decompressing the ubuntu .zip archive, it patches the installer, starts the dependency installation which was already done beforehand so there is nothing to install or upgrade:
```
dkms is already the newest version (2.8.4-3).
libdrm-dev is already the newest version (2.4.104-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
but then it fails with the following error message:
```
Unsatisfied dependencies. Missing component: DKMS.
This is a fatal error, cannot install DisplayLink Linux Software.
```
It is quite puzzling as DKMS is installed. I could not find any way to continue from this. The debug information is as follows:
```
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-5-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.0+dfsg
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Co.,
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Oland [Radeon HD 8570 / R5 430 OEM / R7 240/340 / Radeon 520 OEM] (rev 87)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x53 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 2 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:25:00.0

-------------------------------------------------------------------
```

Please let me know if there is further information needed for investigating this.

Barabazs, 2022-01-06:
Can you post the complete output when you try to install it please?

nullquine, 2022-01-06:
of course:

```
root@Theseus:~/displaylink-debian# ./displaylink-debian.sh 

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.10.0-5-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4036?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2022-01-06 18:50:45--  https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 34.214.248.103, 52.41.97.53
Connecting to www.synaptics.com (www.synaptics.com)|34.214.248.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14500327 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.1.zip’

DisplayLink_Ubuntu_5.4.1.zip        100%[===================================================================>]  13.83M  4.92MB/s    in 2.8s    

2022-01-06 18:50:49 (4.92 MB/s) - ‘DisplayLink_Ubuntu_5.4.1.zip’ saved [14500327/14500327]

Removing prior: "5.4.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.1.zip
  inflating: 5.4.1/displaylink-driver-5.4.1-55.174.run  
Creating directory displaylink-driver-5.4.1-55.174
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.1-55.174  100%  
patching file 5.4.1/displaylink-driver-5.4.1/displaylink-installer.sh
Hunk #1 succeeded at 456 (offset -1 lines).

Installing driver version: 5.4.1

DisplayLink Linux Software 5.4.1-55.174 install script called: install
Distribution discovered: Debian GNU/Linux 11 (bullseye)
[ Installing dependencies ]

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
dkms is already the newest version (2.8.4-3).
libdrm-dev is already the newest version (2.4.104-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Do you want to continue? [Y/n] y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.4-3).
libdrm-dev is already the newest version (2.4.104-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Unsatisfied dependencies. Missing component: DKMS.
This is a fatal error, cannot install DisplayLink Linux Software.

```

and that's all. the debug info  is the same as above.

I booted the system with the device (i-tec U3DVIDOCK) plugged in, it shows up behind a VLI USB3 hub, not sure if that's relevant.

perhaps it is a version conflict?

nullquine, 2022-01-06:
As I see this is a recurring issue but from what I found from the similar bug reports, there is no one definitive cause and therefore no universal solution. Most of the clues I found are hinting at version conflicts.

nullquine, 2022-01-29:
Found the probable cause: "The supported linux kernel version range is from 4.15 to 5.5" from the official website. Likely a version conflict, as I attempted to install it to a system with kernel version 5.10, so the issue is likely not with the wrapper but the vendor-supplied binary itself. Closing for now.

closed 2022-01-29 by nullquine