AdnanHodzic/displaylink-debian #671: Will there be a support for the kernel 5.15 fo kali?

melvintj, 2022-01-11:
┌──(xxxxx㉿kali)-[~/Programs/displaylink_dockingstation/displaylink-debian-master]
└─$ sudo ./displaylink-debian.sh --debug                                                                                                                                                                                                

---------------------------------------------------------------

Unsuported kernel version: 5.15
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

                                                                                                                                                                                                                                              
┌──(xxxxx㉿kali)-[~/Programs/displaylink_dockingstation/displaylink-debian-master]
└─$ lsb_release -a                                                                                                                                                                                                                        
No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2021.4
Codename:	kali-rolling

ArturEPinheiro777, 2022-01-15:
same issue here

SantiagoHoxton79, 2022-01-19:
I, too, am experiencing the same problem on several DEs of Kali (xfce, kde) and Parrot OS. A fix would be greatly appreciated.

Barabazs, 2022-01-30:
Right now it's unsupported by DisplayLink, unless you install their beta version.
I'm not planning on supporting a beta release.

closed 2022-01-30 by Barabazs