AdnanHodzic/displaylink-debian #676: [Platform Request] Linux Mint "Una"

kevinpthorne, 2022-01-14:
I do not think there was any major changes from Uma to Una. Correct me if I am wrong.

ruudbeukema, 2022-01-19:
I thought so too but while the displaylink-driver was working for me on Uma it isn't anymore on Una. After uninstalling and then installing the displaylink-debian driver again, the debug output is as follows:

```bash
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: una
Kernel: 5.11.0-46-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c1)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

On Uma, the debug-output still listed version 1.9.1 if I recalled correctly. Now `sudo modinfo evdi` says that there's no evdi module indeed.

I subsequently installed evdi-1.9.1 from the package manager and rebooted. Still no working external monitors. Debug-output now is:

```bash
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: una
Kernel: 5.11.0-46-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c1)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

Any suggestions on how I can investigate this problem?

ruudbeukema, 2022-01-19:
In order to be able to post the install log, I uninstalled the displaylink-debian driver again and then installed it again (still with the evdi module present from the package manager). The output is as folows:

```bash
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-46-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4036?filetype=exe
Do you accept?: [Y/n]: 

Downloading DisplayLink Ubuntu driver:

--2022-01-19 15:09:24--  https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 34.214.47.153, 34.214.248.103
Connecting to www.synaptics.com (www.synaptics.com)|34.214.47.153|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14500327 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.1.zip’

DisplayLink_Ubuntu_5.4.1.zip                                100%[=========================================================================================================================================>]  13,83M  3,50MB/s    in 4,7s    

2022-01-19 15:09:30 (2,96 MB/s) - ‘DisplayLink_Ubuntu_5.4.1.zip’ saved [14500327/14500327]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.1.zip
  inflating: 5.4.1/displaylink-driver-5.4.1-55.174.run  
Creating directory displaylink-driver-5.4.1-55.174
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.1-55.174  100%  
patching file 5.4.1/displaylink-driver-5.4.1/displaylink-installer.sh
Hunk #1 succeeded at 456 (offset -1 lines).

[... 86 lines cut ...]
X11 version: 1.20.13-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

ruudbeukema, 2022-01-29:
Based on the above situation I removed the `20-displaylink.conf` file and rebooted, but it didn't solve the problem.
I then upgraded to the `5.13.0-27-generic` kernel. This DID result in the Linux Mint desktop being shown on the two external monitors as well. `glxgears` still reports 1FPS for those external monitors, but that was also the case with LM Uma and is caused by the AMD driver I believe. Or maybe I should now try with the `20-displaylink.conf` file back in place.

JabWik, 2022-02-03:
Hello, I disabled secure boot as mentioned in the guide and checked if I work on Xserver, so I meet all the requirements.

`--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.13.0-28-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

---------------------------------------------------------------

Unsuported platform: Linuxmint una
Full list of all supported platforms: http://bit.ly/2zrwz2u

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: una
Kernel: 5.13.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9a49 (rev 01)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

The driver didn't even install because when I type in terminal `sudo displaylink-debian.sh --debug``
`sudo: displaylink-debian.sh: command not found`, 
moreover there is no running status for DisplayLink.

PR #682 (2022-02-03): Add support for Linux Mint 20.3 Una
Closes #676
closed 2022-02-03 by Barabazs