AdnanHodzic/displaylink-debian #677: When second monitor connected the boot process gets broken and any change in display config freeze screens

kamo93, 2022-01-17:
Hello,

I having an issue where I can start my pc if the second monitor is connected and anytime I change something in the display settings the screen just freeze and I need to make hard reboot.

This is my `sudo ./displaylink-debian.sh --debug` result:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.11.0-46-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation UHD Graphics 630 (Mobile)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Max-Q] (rev a1)
X11 version: 1.20.13-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2d2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2b0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x28e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x26c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1dd cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting

Also have a stupid question why it says in version 1.9.1 shouldn't be 5.4.xx 

Thanks in advance

mcoyle1960, 2022-01-20:
There is a beta driver here which fixed that problem for me.

https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

kamo93, 2022-01-20:
@mcoyle1960 Thanks for your help. Sorry I'm really new with linux/ubuntu so I not sure how to install the beta version. Do I need to install it through displaylink-debian.sh or just download and install it independently.

mcoyle1960, 2022-01-20:
Run the **displaylink-driver-5.5.0-beta-59.118.run** script. You don't use the script from this site. It doesn't look for beta versions. Good luck!

kamo93, 2022-01-24:
@mcoyle1960 Thanks works perfect!

closed 2022-01-24 by kamo93