AdnanHodzic/displaylink-debian #678: DisplayLink not working on debian Bullseye

Eristox, 2022-01-18:
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation CometLake-S GT2 [UHD Graphics 630] (rev 03)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2022-02-03:
What's not working for you? Please provide more details.

Eristox, 2022-02-03:
Hello, thank you for your feedback

I work on a computer with only 2 video outputs.
But I need three monitors, I opted for a USB/HDMI adapter for the third monitor. 

Impossible to make it work, I followed the displaylink procedure for debian bulleyes but it doesn't work I can't use my third screen. It does not seem to be detected by xrandr or arandr

Barabazs, 2022-02-03:
Which adapter is it?

Eristox, 2022-02-03:
No mark on the adapter. 
But with an lsusb I get MacroSilicon VGA Display Adapter
But it is a USB/HDMI adapter

Barabazs, 2022-02-03:
Are you sure it's a DisplayLink compatible adapter?
What's the chipset?

artukryp, 2022-02-05:
I saw something curious, not sure if you already know about it. I find that If I use lightdm as display manager, the image does not appear but if I change the display manager to gdm3 and then I install DisplayLink, the usb extern monitors works good just in the first logging screen:
[image]

closed 2022-03-12 by Barabazs