AdnanHodzic/displaylink-debian #681: Ubuntu 22.04 support

aghand0ur, 2022-01-31:
@AdnanHodzic 
Unsuported kernel version: 5.15
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

Barabazs, 2022-01-31:
Kernel 5.15 is not supported by DisplayLink stable version.
I'm not planning on supporting a beta release...

closed 2022-01-31 by Barabazs

aghand0ur, 2022-02-02:
It would be nice to add a bash option to fetch beta drivers if the user is willing to take the risk

Barabazs, 2022-02-02:
Just downloading the beta driver is not hard, but it's all the subsequent handling of the file that requires more work.

aghand0ur, 2022-02-02:
i see

aghand0ur, 2022-03-12:
@Barabazs @AdnanHodzic 
I assume Kernel 5.15 is now  supported by DisplayLink stable version per below link:
https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

Any plan to merge it in your too?

Barabazs, 2022-03-12:
Hi @aghand0ur, I'm working on it right now. It should be ready later today.

aghand0ur, 2022-03-12:
Excellent news. Thank you @Barabazs