AdnanHodzic/displaylink-debian #683: Is possible update for kali 2021.4a? Thank you very much.

ubercook, 2022-02-03:


Barabazs, 2022-02-04:
Please post the output of `sudo ./displaylink-debian.sh --debug`

artukryp, 2022-02-05:
Unsuported kernel version: 5.14
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

rogue73, 2022-02-20:
Hi, 
may be you are able to install with my forked repository where I've add some changes.  

https://github.com/rogue73/displaylink-debian

I've done these changes to get it working on my laptop with Debian/sid that was just updated to kernel 5.16.
kali is based on debian so maybe you can successful install displaylink driver as well.

The changes I've done are:
- allow installation up to kernel 5.16
- download and use always the latest displaylink driver even if it's beta
- use a different patch file to patch the original installer script because there are some changes and the old patchfile does not match

After these changes i could install die driver on my laptop with debian/sid kernel 5.16 and connected to a itec displaylink dock. I have no experience how stable it is. Currently it is working well.

Cheers,
Rogue73

artukryp, 2022-02-28:
I tested with your changes. Using kernel 5.15 and after the installation I lost the graphics I just can use terminals, I tried to uninstall but appears is not posible.

rogue73, 2022-02-28:
Did you have a look at post-install-guide.md? Maybe there are some steps that needs to be done to activate your displays.  
At my installation for example i had to activate the providers
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0

I haven`t tried the uninstall after i could successful install the beta version but i will try it as soon as possible and give you an update.

dsbrgg, 2022-03-07:
> Hi, may be you are able to install with my forked repository where I've add some changes.
> 
> [rogue73/displaylink-debian](https://github.com/rogue73/displaylink-debian)
> 
> I've done these changes to get it working on my laptop with Debian/sid that was just updated to kernel 5.16. kali is based on debian so maybe you can successful install displaylink driver as well.
> 
> The changes I've done are:
> 
> * allow installation up to kernel 5.16
> * download and use always the latest displaylink driver even if it's beta
> * use a different patch file to patch the original installer script because there are some changes and the old patchfile does not match
> 
> After these changes i could install die driver on my laptop with debian/sid kernel 5.16 and connected to a itec displaylink dock. I have no experience how stable it is. Currently it is working well.
> 
> Cheers, Rogue73

Just wanted to point out that this fork worked perfectly for me on Pop!_OS 21.10, just had to [remove this section](https://github.com/rogue73/displaylink-debian/blob/master/displaylink-debian.sh#L703) as this was causing my vertical monitor to freeze or the display would turn black even though it was drawing the mouse movement and in my case I'm using a NVIDIA GPU which is already defined in [this device section](https://github.com/rogue73/displaylink-debian/blob/master/displaylink-debian.sh#L714) a couple of line below.