AdnanHodzic/displaylink-debian #684: Displays are not displaying anything on Ubuntu 20

kaangoksal, 2022-02-05:
here is my debug output

```
root@ubuntu-display-test:/home/ubuntu/displaylink-debian# displaylink-installer --debug
DisplayLink Linux Software 5.4.1-55.174 install script called: --debug
Distribution discovered: Ubuntu 20.04.3 LTS

Installs DisplayLink Linux Software, version 5.4.1-55.174.
Usage: /usr/bin/displaylink-installer [ install | uninstall ]

The default operation is install.
If unknown argument is given, a quick compatibility check is performed but nothing is installed.
root@ubuntu-display-test:/home/ubuntu/displaylink-debian# ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.13.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: bochs-drm
Subsystem: Device
VGA: Device 1234:1111 (rev 02)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x41 cap: 0x0 crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 1: id: 0xe9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xc7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xa5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x7b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

Troubleshooting steps that are tried; 

1) adding a  Xorg configuration file at `/usr/share/X11/xorg.conf.d`

as following: 
```
ubuntu@ubuntu-display-test:~$ cat 20-evidevice.conf 
Section "OutputClass"
        Identifier "DisplayLink"
        MatchDriver "evdi"
        Driver "modesetting"
        Option  "AccelMethod" "none"
EndSection

```

kaangoksal, 2022-02-06:
Just tried the same configuration with a Windows 10 Guest. It works like a charm. No latency, can even watch 1080p youtube videos. This would be amazing if we can sort it out for Linux as well.

kaangoksal, 2022-02-06:
The problem is with linux display drivers, they don't have a source output 
```
ubuntu@ubuntu-display-test:~$ xrandr --listproviders
Providers: number : 5
Provider 0: id: 0x47 cap: 0x0 crtcs: 4 outputs: 4 associated providers: 0 name:qxl
Provider 1: id: 0xfc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xda cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xb8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x8f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

My physical graphics card has source output 
```
$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 7 associated providers: 0 name:NVIDIA-0

```

Barabazs, 2022-03-12:
Hi @kaangoksal. Your question is unrelated to displaylink-debian and is more appropriate for a forum/stackoverflow concerning xorg configurations.

closed 2022-03-12 by Barabazs

tristan-k, 2022-04-02:
I'm in the [same boat](https://www.reddit.com/r/Proxmox/comments/slpthq/comment/i33gemv/?utm_source=reddit&utm_medium=web2x&context=3) as @kaangoksal. Did you find a solution?

kaangoksal, 2022-04-02:
@tristan-k 

I couldn't, a user on reddit said it worked with Wayland, but I had no success with that. 
https://www.reddit.com/r/Proxmox/comments/slpthq/physical_video_outputs_for_proxmox_guests_using/hwi8w91/?context=3

what the user said: 
```
I have this working using a Wavlink USB 3.0 to HDMI adaptor. I had to use a Wayland session, then enable autologin and disable the virtual display from proxmox (detected as Red Hat, Inc. QEMU Monitor) in Display configuration on Manjaro KDE
```

When I tried this, the following happened: 

1) I switched the default session to wayland
2) After displaylink drivers got installed, my default session became x11 and I couldn't revert it back to wayland. 

end of story.