AdnanHodzic/displaylink-debian #685: usb mouse and keyboard lag

marionatiello, 2022-02-07:
Hi, I have a second screen connected via a Dell universal dock D6000 to a dell laptop running debian 11.
The installation of displaylink worked fine. However, when I close the lid (or equivalently if I issue 
xrandr --output eDP-1 --off
to turn the laptop screen off) the usb mouse & keyboard (connected to a usbport on the dock) lag notoriously.
If I issue xrandr --auto the lag stops but of course the laptop screen is not off.

Is this a xrandr issue or does it relate to the displaylink-debian software?
Thanks for helping
MNatiello

MNatiello, 2022-02-22:
Hi, I posted the "ghost" issue here. 
I run debian 11 bullseye. Displaylink discovers and handles my external monitor with two issues: one minor and one major.
The major: when I put the laptop screen off usb-mouse and usb-monitor lag several seconds. Both closing the lid and issuing xrandr --output eDP-1 --off display the same behaviour.
usb mouse and monitor are at 
Bus 001 Device 008: ID 046d:c52b Logitech, Inc. Unifying Receiver
There is no clue in Xorg.0.log or dmesg or kern.log

The minor issue: Every 30 minutes or so, the external screen goes black and the keyboard and mouse freeze for 10-15 seconds. After that they recover and the external screen is on again.

How should I proceed?
M. Natiello

Barabazs, 2022-03-12:
The lid-closing bug is a known issue and not related to displaylink-debian.

closed 2022-03-12 by Barabazs

MNatiello, 2022-03-12:
Hi

> The lid-closing bug is a known issue and not related to
> displaylink-debian.

Well, that is not crystal clear. I connected my external
monitor to the DP-1 and DP-2 ports of the laptop and there
are no problems whatsoever. Neither mouse issues when
closing the lid, nor flicker or flashes on the external
monitor. For the time being I am not using display link
to connect my external monitor.

Yours,

M. Natiello

-- 
***@***.***     
Professor in Applied Mathematics, Dr., Docent, ETP
Centre for Mathematical Sciences, Lund University
Box 118, S-221 00 LUND, SWEDEN Tel: +46-46-2220919  
http://www.maths.lu.se/staff/mario-natiello
--
When you send emails to Lund University, we process your personal
data in accordance with existing legislation. To find out more about the
processing of your personal data, visit Lund University's website,
www.lunduniversity.lu.se/about/contact-us/processing-of-personal-data-at-lund-university

Barabazs, 2022-03-12:
Hi @MNatiello. What I mean, is that your issue is not related to displaylink-debian (which is only a script to help install DisplayLink). Your issue _is_ related to DisplayLink or evdi, but there is nothing we can do for you here.
There are a few issues in their repo that mention the lid bug. Unfortunately most were never fixed. 
https://github.com/DisplayLink/evdi/issues?q=is%3Aissue+lid