AdnanHodzic/displaylink-debian #686: Add support for Kali Linux 2022.1

Aotricx, 2022-02-08:
If you do you can add me on discord and I will send you $20 on cashapp or zelle for your work.
Discord: Kunaii#8179

Barabazs, 2022-02-28:
Please post the output of `sudo ./displaylink-debian.sh --debug`

PwnPeter, 2022-03-03:
Hire, below the output :

```bash
---------------------------------------------------------------
─$ sudo ./displaylink-debian.sh --debug

---------------------------------------------------------------

Unsuported kernel version: 5.16
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

```

Aotricx, 2022-03-03:
hey sorry i havent been home, i should be able to do it tonight

Balhech, 2022-03-09:
---------------------------------------------------------------

Unsuported kernel version: 5.16
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

Barabazs, 2022-03-12:
`displaylink-debian` has now been updated.
Please try with the latest changes.

closed 2022-03-12 by Barabazs