AdnanHodzic/displaylink-debian #691: New DisplayLink Release [5.5] breaks strategy displaylink-debian.sh script

sumadark, 2022-03-10:
Hello,

I am actually trying to install displaylink-debian on my Debian 11, and I am facing multiple errors.
First is the same as mentionned in this issue : [https://github.com/AdnanHodzic/displaylink-debian/issues/662](url) 

I was able to change the script to be able to pass this issue, but then I am encountering this : 
`DisplayLink_Ubuntu_5.5.zip     100%[====================================================>]  16,11M  4,50MB/s    ds 3,6s    

2022-03-10 11:41:32 (4,50 MB/s) — « DisplayLink_Ubuntu_5.5.zip » sauvegardé [16893957/16893957]

Removing prior: "5.5" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.5.zip
  inflating: 5.5/DisplayLink USB Graphics Software for Ubuntu 5.5.0-59.151-Release Notes.txt  
  inflating: 5.5/displaylink-driver-5.5.0-59.151.run  
Creating directory displaylink-driver-5.5.0-59.151
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.5.0-59.151  100%  
patching file 5.5/displaylink-driver-5.5/displaylink-installer.sh
Hunk #1 FAILED at 457.
1 out of 1 hunk FAILED -- saving rejects to file 5.5/displaylink-driver-5.5/displaylink-installer.sh.rej`

Here is the content of "displaylink-installer.sh.rej" : 

`--- /opt/displaylink/displaylink-installer.sh
+++ /opt/displaylink/displaylink-installer.sh
@@ -457,11 +457,11 @@
   echo "http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu"
 
   echo -e "\nInstallation complete!"
-  echo -e "\nPlease reboot your computer if intending to use Xorg."
-  $XORG_RUNNING || exit 0
-  read -rp 'Xorg is running. Do you want to reboot now? (Y/n)' CHOICE
-  [[ ${CHOICE:-Y} =~ ^[Nn]$ ]] && exit 0
-  reboot
+  #echo -e "\nPlease reboot your computer if intending to use Xorg."
+  #$XORG_RUNNING || exit 0
+  #read -rp 'Xorg is running. Do you want to reboot now? (Y/n)' CHOICE
+  #[[ ${CHOICE:-Y} =~ ^[Nn]$ ]] && exit 0
+  #reboot
 }
 
 uninstall()`

Thanks for helping me with this :-)

Suma

Zonkil9, 2022-03-10:
Okay, I did a little tweak to the code. Please follow the steps.

1. Clone the repository again.
2. Open displaylink-debian.sh in any text editor.
3. Find the line 
```
driver_dir=$version
```
4. BEFORE that line insert three new lines:
```
version=5.4.1
dlurl=https://www.synaptics.com/node/4036?filetype=exe
driver_url=https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
```
5. It should look like:
```
[...]
fi
version=5.4.1
dlurl=https://www.synaptics.com/node/4036?filetype=exe
driver_url=https://www.synaptics.com/sites/default/files/exe_files/2021-09/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4.1-EXE.zip
driver_dir=$version
[...]
```

Now the script is forced to install version older version 5.4.1. It is just a temporary solution if you need to make it work right now.

sumadark, 2022-03-10:
Thank you for your quick answer, I will try this tomorrow.
I was able to install the version 5.5 directly from the synaptics website, and it seems to work but it isn't very smooth.

cilynx, 2022-03-11:
Tweaks to make the new upstream 5.5 work with this repo.  Confirmed working on Debian Sid running Linux version 5.16.0-3-amd64.  I'm seeing a little bit of tearing on the pointer, but overall it's working well.

https://github.com/cilynx/displaylink-debian/commits/upstream-5.5

drGonzoBdm, 2022-03-11:
> Tweaks to make the new upstream 5.5 work with this repo. Confirmed working on Debian Sid running Linux version 5.16.0-3-amd64. I'm seeing a little bit of tearing on the pointer, but overall it's working well.
> 
> https://github.com/cilynx/displaylink-debian/commits/upstream-5.5

Hello @cilynx and thank you for this solution.  

It works fine for the installation but I have a big latency problem ( the mouse pointer is moving almost each scond on the screen plugged on the station ) as soon I use my docking station.  

I recently received my dell d3100 and I was hoping that this new release should solve the problem.  
I have look in most of the opened issues without finding a solution.  

Maybe the full AMD configuration ( Ryzen 7 and AMD Radeon Vega 10 Graphics ) ?  

If the creation of a new issue is needed tell me :)

Barabazs, 2022-03-12:
`displaylink-debian` has now been updated.
Please try with the latest changes.

closed 2022-03-12 by Barabazs