AdnanHodzic/displaylink-debian #698: [Linux Mint 20.3] Unbearable lag + screen tearing

evprkr, 2022-03-19:
Got everything install and "working" on a fresh install of Mint, but the latency, lag, and screen tearing on the monitor makes the whole thing almost entirely pointless. It's bad enough that it's difficult to even use the monitor to write code (which is the entire reason I use this one, it's my portrait monitor)

I've been googling around, haven't found any fixes that worked out. Is there anything that can be done about this? A lot of threads I found about this issue in Ubuntu said switching to wayland fixed the problem, but I don't believe I have that option on Mint.

Thanks.

evprkr, 2022-03-19:
I also want to note that I have followed the post-install guide, I was required to delete the conf file and reboot in order to get the monitor to work at all.

AlexDaniel, 2022-03-22:
First time running it, so I'm not sure if it's a regression or not, but I'm seeing unbearable lag on debian unstable.

Barabazs, 2022-04-19:
I don't think there's anything displaylink-debian can do here. Your best option is to make a post on the DisplayLink forums.

closed 2022-04-19 by Barabazs