AdnanHodzic/displaylink-debian #701: Debian 11 can't install

Helphyy, 2022-03-24:
`unzip is installed
dpkg-query: le paquet « linux-headers-5.10.0-11-amd64 » n'est pas installé et aucune information n'est disponible
Use dpkg --info (= dpkg-deb --info) to examine archive files.
linux-headers-5.10.0-11-amd64 not found! Install? [Y/n] y
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
E: Impossible de trouver le paquet linux-headers-5.10.0-11-amd64
E: Impossible de trouver de paquet correspondant à l'expression rationnelle « linux-headers-5.10.0-11-amd64 »
E: Impossible de trouver de paquet correspondant à l'expression rationnelle « linux-headers-5.10.0-11-amd64 »
linux-headers-5.10.0-11-amd64 installation failed.  Aborting.`

here is my debug :

`--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation WhiskeyLake-U GT2 [UHD Graphics 620] (rev 02)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX250] (rev a1)
X11 version: 1.20.11-1+deb11u1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

thanks you very much

Barabazs, 2022-03-25:
Can you post the output of
`apt search linux-source`
`apt show linux-source`
`apt search linux-source-$(uname -r)`
`apt show linux-source-$(uname -r)`

Helphyy, 2022-03-26:
@Barabazs how can i a label btw ?

Barabazs, 2022-03-27:
I'm not a debian user myself, so I'm not 100% sure, but it looks like you should try using a newer kernel version. At least version 5.10.103-1.

> @Barabazs how can i a label btw ?

I don't think you can do without special privileges.

depitropov, 2022-04-01:
As far as I remember I installed a newer version of linux-kbuild from buster backports (first you need to add buster-backports to aptsource). 

`sudo apt-get install linux-kbuild-5.10/buster-backports`

linux-headers-5.10.0-11-amd64 required a newer version of linux-kbuild that was not in the stable bullseye repository.

too-difficult-to-name, 2022-04-04:
> 你可以发布的输出 `apt search linux-source` `apt show linux-source` `apt search linux-source-$(uname -r)` `apt show linux-source-$(uname -r)`

i 'm deepin too,this is a message for debug:

closed 2022-04-19 by Barabazs