AdnanHodzic/displaylink-debian #702: Installation generates corrupted 20-displaylink.conf file (Device(s) detected but none match those in the config files)

depitropov, 2022-04-01:
1. I've installed Nvidia drivers
2. I've installed display-link-debian
3. I've signed evdi module
4. 
During boot there is a problem with 20-displaylink.conf configuration. Xorg cannot start and the following is in the log:

```
(II) modeset(0) using default device
(EE) Screen 0 deleted because of not matching config section
(II) UnloadModuke: "modsetting"
(EE) Device(s) detected but none match those in the config files
EE) no screens found(EE)
```

After deleting 20-displaylink.conf Xorg successfully loads and DisplayLink works, but I get random screen disconnects from time to time.

20-displaylink.conf:

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

```

Debug output:

```

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-12-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.1
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [103c:888d]
VGA: Intel Corporation Device 9a60 (rev 01)
VGA (3D): NVIDIA Corporation Device 1fba (rev a1)
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x49 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x337 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x315 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2f3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2d1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x2af cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0

-------------------------------------------------------------------
```

Barabazs, 2022-04-19:
Does the file /etc/X11/20-displaylink.conf still exist or is that the one you deleted?
If it exists, please post the content.

closed 2022-06-01 by Barabazs

depitropov, 2022-09-26:
@Barabazs Hey was my issue a bug that was solved in a newer version?
I didn't get why the issue was closed.

Thanks!