AdnanHodzic/displaylink-debian #704: Displaylink Monitor screen blank. Nvidia Driver, etc

arturogatti, 2022-04-02:
$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.0
1.10.1
1.9.0+dfsg
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: [NVS
GT218
VGA: NVIDIA Corporation GF119 [NVS 310] (rev a1)
NVIDIA Corporation GT218 [GeForce 210] (rev a2)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/20-intel.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x43 cap: 0x0 crtcs: 2 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xbe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x9c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x6f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

arturogatti, 2022-04-02:
after install its always

$ sudo systemctl start dlm.service
Failed to start dlm.service: Unit dlm.service not found.

arturogatti, 2022-04-02:
$ xrandr --listproviders
Providers: number : 5
Provider 0: id: 0x43 cap: 0x0 crtcs: 2 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xbe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x9c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x6f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

arturogatti, 2022-04-02:
sudo dmesg | grep Display
[    2.356669] usb 2-10: Manufacturer: DisplayLink
[   55.152164] evdi: [I] (card0) Opened by Task 1456 (DesktopManagerE) of process 1356 (DisplayLinkMana)
[   55.432615] evdi: [I] (card0) Connected with Task 1456 (DesktopManagerE) of process 1356 (DisplayLinkMana)

arturogatti, 2022-04-02:
$systemctl status dlm.service failure
Unit dlm.service could not be found.
Unit failure.service could not be found.

arturogatti, 2022-04-02:
$ lspci | grep VGA
02:00.0 VGA compatible controller: NVIDIA Corporation GF119 [NVS 310] (rev a1)
03:00.0 VGA compatible controller: NVIDIA Corporation GT218 [GeForce 210] (rev a2)

arturogatti, 2022-04-02:
im about to switch to fedora I guess.

Barabazs, 2022-04-04:
You have to associate the providers, as noted in the [post-install-guide#display-detection](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection).

Barabazs, 2022-04-19:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2022-04-19 by Barabazs

arturogatti, 2022-04-20:
Respectfully, can someone be of a little more help.? A referral to the debug guide that myself and hundreds of others saw hundreds of times while having to uninstall + reinstall this  debian displaylink driver / make shift patch doesn't do much. I wouldnt be here if it did. 
And the link you referred me to in the guide just assumes that everyone automatically knows what " <prov-xid> <source-xid>" is referring to. Nobody knows what that even is dude. 

Getting "Minor opcode of failed request:  35 (RRSetProviderOutputSource)" errors. See below.

Direct answers are appreciated.

xrandr --setprovideroutputsource 2 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource) 
  Value in failed request:  0x43
  Serial number of failed request:  19
  Current serial number in output stream:  20

X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x43
  Serial number of failed request:  19
  Current serial number in output stream:  20

arturogatti, 2022-04-20:
plz reopen this issue...