AdnanHodzic/displaylink-debian #705: missing xorg.conf.d etc

davecman, 2022-04-06:
I  have been trying for many hours to get DisplayLink to install on my brand new computer, B550-F motherboard, 5900-X CPU and Zorin. Lots of errors, depending on the version of displaylink I try to install. Errors include:

  Building EVDI kernel module with DKMS
  Failed to build evdi/5.2.14

  make -j24 KERNELREL....
  Failed to install evdi to the kernel tree.

error: evdi-dkms
Sub-process /usr/bin/dpkg returned an error code (1)

I finally found displaylink-debian and followed those instructions, failed, then ran debug with the following output:

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Zorin
Release: focal
Kernel: 5.13.0-39-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: 
Subsystem: Ltd
VGA: NVIDIA Corporation GK208B [GeForce GT 730] (rev a1)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

Are you able to help me please?

Thank you.

Dave Clingman

Barabazs, 2022-04-06:
Did you try to install with displaylink-debian or with the official package?

davecman, 2022-04-06:
I've tried both. Three versions of the full package plus the debian pack.

davecman, 2022-04-13:
I think I have finally figured out that my system simply does not have EVDI installed, and I can't figure out how to do it. I followed instructions from: 
https://forums.developer.nvidia.com/t/building-displaylink-evdi/73626
But when I run the "sudo dkms build evdi/devel command, I get an error:

make -j24 KERNELRELEASE=5.13.0-39-generic all INCLUDEDIR=/lib/modules/5.13.0-39-generic/build/include KVERSION=5.13.0-39-generic DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.13.0-39-generic (x86_64)
Consult /var/lib/dkms/evdi/devel/build/make.log for more information.

I have spent more than 10 hours trying to get Displaylink installed and am beyond frustrated.

Please someone help.

Thanks.

davecman, 2022-04-13:
lsmod returns:

Module                  Size  Used by
rfcomm                 81920  4
ccm                    20480  6
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   24576  2
nls_iso8859_1          16384  1
intel_rapl_msr         20480  0
intel_rapl_common      24576  1 intel_rapl_msr
edac_mce_amd           32768  0
snd_hda_codec_realtek   147456  1
snd_hda_codec_generic    81920  1 snd_hda_codec_realtek
ledtrig_audio          16384  1 snd_hda_codec_generic
kvm                   864256  0
btusb                  61440  0
snd_hda_codec_hdmi     61440  1
btrtl                  24576  1 btusb
btbcm                  20480  1 btusb
btintel                32768  1 btusb
crct10dif_pclmul       16384  1
ghash_clmulni_intel    16384  0
bluetooth             651264  31 btrtl,btintel,btbcm,bnep,btusb,rfcomm
aesni_intel           376832  8
snd_hda_intel          53248  5
mt7921e                77824  0
snd_intel_dspcfg       28672  1 snd_hda_intel
crypto_simd            16384  1 aesni_intel
mt76_connac_lib        36864  1 mt7921e
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel
snd_hda_codec         147456  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
mt76                   77824  2 mt7921e,mt76_connac_lib
ecdh_generic           16384  1 bluetooth
snd_seq_midi           20480  0
rapl                   20480  0
joydev                 28672  0
input_leds             16384  0
eeepc_wmi              16384  0
ecc                    36864  1 ecdh_generic
wmi_bmof               16384  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_hda_core           94208  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
mac80211             1028096  3 mt76,mt7921e,mt76_connac_lib
snd_rawmidi            36864  1 snd_seq_midi
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               114688  5 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
snd_seq                73728  2 snd_seq_midi,snd_seq_midi_event
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
ccp                    98304  0
snd_timer              40960  2 snd_seq,snd_pcm
efi_pstore             16384  0
snd                    94208  20 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
cfg80211              888832  4 mt76,mt7921e,mac80211,mt76_connac_lib
k10temp                16384  0
soundcore              16384  1 snd
libarc4                16384  1 mac80211
mac_hid                16384  0
sch_fq_codel           20480  5
ipmi_devintf           20480  0
ipmi_msghandler       114688  1 ipmi_devintf
msr                    16384  0
parport_pc             45056  0
ppdev                  24576  0
lp                     20480  0
parport                65536  3 parport_pc,lp,ppdev
ip_tables              32768  0
x_tables               49152  1 ip_tables
autofs4                45056  2
hid_generic            16384  0
usbhid                 57344  0
hid                   139264  2 usbhid,hid_generic
nouveau              2064384  70
mxm_wmi                16384  1 nouveau
i2c_algo_bit           16384  1 nouveau
drm_ttm_helper         16384  1 nouveau
ttm                    69632  2 drm_ttm_helper,nouveau
mfd_aaeon              16384  0
drm_kms_helper        253952  1 nouveau
asus_wmi               36864  2 eeepc_wmi,mfd_aaeon
sparse_keymap          16384  1 asus_wmi
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
cec                    53248  1 drm_kms_helper
rc_core                61440  1 cec
drm                   557056  21 drm_kms_helper,drm_ttm_helper,ttm,nouveau
video                  53248  2 asus_wmi,nouveau
crc32_pclmul           16384  0
nvme                   49152  2
xhci_pci               24576  0
igc                   122880  0
ahci                   40960  0
gpio_amdpt             20480  0
i2c_piix4              28672  0
nvme_core             126976  3 nvme
xhci_pci_renesas       20480  1 xhci_pci
libahci                36864  1 ahci
wmi                    32768  5 asus_wmi,wmi_bmof,mfd_aaeon,mxm_wmi,nouveau
gpio_generic           20480  1 gpio_amdpt

davecman, 2022-04-19:
Can seriously no one help me at all?

Barabazs, 2022-04-19:
Mixing different installation methods won't do you any good. This might even make it harder to troubleshoot...
Can you try the steps below and report please? 

1. `sudo ./displaylink-debian.sh --install`
2. Copy the output of the installation and save it somewhere.
3. Reboot and see if it works.
4. Post the output of the installation here.

davecman, 2022-04-19:
Thank you for taking the time to provide clear instructions. Same error as every single time with every single distro I've tried. Output follows:

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.13.0-39-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4391?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2022-04-19 17:27:40--  https://www.synaptics.com/sites/default/files/exe_files/2022-03/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.5-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 52.43.2.25, 35.167.86.240
Connecting to www.synaptics.com (www.synaptics.com)|52.43.2.25|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16893957 (16M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.5.zip’

DisplayLink_Ubuntu_ 100%[===================>]  16.11M  1.12MB/s    in 19s     

2022-04-19 17:28:00 (887 KB/s) - ‘DisplayLink_Ubuntu_5.5.zip’ saved [16893957/16893957]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.5.zip
  inflating: 5.5/DisplayLink USB Graphics Software for Ubuntu 5.5.0-59.151-Release Notes.txt  
  inflating: 5.5/displaylink-driver-5.5.0-59.151.run  
Creating directory displaylink-driver-5.5.0-59.151
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.5.0-59.151  100%  
patching file 5.5/displaylink-driver-5.5/displaylink-installer.sh
patch unexpectedly ends in middle of line

Installing driver version: 5.5

Distribution discovered: Zorin OS 16.1

Installing
[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.10.1/source ->
                 /usr/src/evdi-1.10.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j24 KERNELRELEASE=5.13.0-39-generic all INCLUDEDIR=/lib/modules/5.13.0-39-generic/build/include KVERSION=5.13.0-39-generic DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.13.0-39-generic (x86_64)
Consult /var/lib/dkms/evdi/1.10.1/build/make.log for more information.

------------------------------
Deleting module version: 1.10.1
completely from the DKMS tree.
------------------------------
Done.
ERROR: Failed to install evdi to the kernel tree.

davecman, 2022-04-19:
I can find no evidence that I have anything EVDI anywhere on my computer, and I have tried various methods of trying to install it, none of which have worked.

ERROR (code 3): Failed to build evdi/4.2.29.

But I am still happy to try anything else you can suggest.

Thank you again for your help, Barabazs.

Barabazs, 2022-04-19:
What is the content of the file `/var/lib/dkms/evdi/1.10.1/build/make.log`

davecman, 2022-04-19:
I have no "1.10.1" folder. I have a dev folder and a devel folder.

Contents of that file in the dev folder:
#
# Copyright (c) 2015 - 2020 DisplayLink (UK) Ltd.
#
# This file is subject to the terms and conditions of the GNU General Public
# License v2. See the file COPYING in the main directory of this archive for
# more details.
#
ifneq ($(DKMS_BUILD),)

# DKMS

KERN_DIR := /lib/modules/$(KERNELRELEASE)/build

ccflags-y := -Iinclude/drm
evdi-y := evdi_drv.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_main.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o
evdi-$(CONFIG_COMPAT) += evdi_ioc32.o
obj-m := evdi.o

KBUILD_VERBOSE ?= 1

all:
	$(MAKE) KBUILD_VERBOSE=$(KBUILD_VERBOSE) M=$(CURDIR) SUBDIRS=$(CURDIR) SRCROOT=$(CURDIR) CONFIG_MODULE_SIG= -C $(KERN_DIR) modules

clean:
	@echo $(KERN_DIR)
	$(MAKE) KBUILD_VERBOSE=$(KBUILD_VERBOSE) M=$(CURDIR) SUBDIRS=$(CURDIR) SRCROOT=$(CURDIR) -C $(KERN_DIR) clean

else

# Not DKMS

ifneq ($(KERNELRELEASE),)

# inside kbuild
# Note: this can be removed once it is in kernel tree and Kconfig is properly used
CONFIG_DRM_EVDI := m
LINUXINCLUDE := $(subst -I,-isystem,$(LINUXINCLUDE))
ccflags-y := -isystem include/drm $(CFLAGS)
evdi-y := evdi_drv.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_main.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o
evdi-$(CONFIG_COMPAT) += evdi_ioc32.o
obj-$(CONFIG_DRM_EVDI) := evdi.o

else

# kbuild against specified or current kernel
CP ?= cp
DKMS ?= dkms
RM ?= rm

MODVER=1.7.0

ifeq ($(KVER),)
	KVER := $(shell uname -r)
endif

ifneq ($(RUN_DEPMOD),)
	DEPMOD := /sbin/depmod -a
else
	DEPMOD := true
endif

ifeq ($(KDIR),)
	KDIR := /lib/modules/$(KVER)/build
endif

MOD_KERNEL_PATH := /kernel/drivers/gpu/drm/evdi

default: module

module:
	$(MAKE) -C $(KDIR) M=$$PWD

clean:
	$(RM) -rf *.o *.a *.ko .tmp* .*.*.cmd Module.symvers evdi.mod.c modules.order

install:
	$(MAKE) -C $(KDIR) M=$$PWD INSTALL_MOD_PATH=$(DESTDIR) INSTALL_MOD_DIR=$(MOD_KERNEL_PATH) modules_install
	$(DEPMOD)

uninstall:
	$(RM) -rf $(DESTDIR)/lib/modules/$(KVER)/$(MOD_KERNEL_PATH)
	$(DEPMOD)

install_dkms:
	$(DKMS) install .

uninstall_dkms:
	$(DKMS) remove evdi/$(MODVER) --all
	$(RM) -rf /usr/src/evdi-$(MODVER)

endif # ifneq ($(KERNELRELEASE),)

endif # ifneq ($(DKMS_BUILD),)

Contents in the devel folder:

DKMS make.log for evdi-devel for kernel 5.13.0-39-generic (x86_64)
Wed 13 Apr 2022 19:33:10 NZST
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/devel/build SUBDIRS=/var/lib/dkms/evdi/devel/build SRCROOT=/var/lib/dkms/evdi/devel/build CONFIG_MODULE_SIG= -C /lib/modules/5.13.0-39-generic/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-39-generic'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:753: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-39-generic'
make: *** [Makefile:33: all] Error 2

Barabazs, 2022-04-19:
> (EDIT: I don't know why some of the text is larger and bold than the rest. In the text editor, it was all the same.)
> 

You have to put the output inside a code block so it doesn't get formatted as markdown. More info [here](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks)
I edited your message with code blocks for clarity.

> I have no "1.10.1" folder. 

Weird, because that's where the installer should have saved the logs.

> I have a dev folder and a devel folder, both of which I created when I tried previously to manually install EVDI.

You made it a lot harder to troubleshoot and almost impossible to find the root cause of your issue.
I would suggest trying to reboot with another kernel version and try installing with displaylink-debian only.

davecman, 2022-04-22:
Thank you, @Barabazs, for all your help.

After backing up everything, I wiped the hard drive and installed Zorin fresh. DisplayLink works fine now, as do both of my inputs on my graphics card, so 4 displays now, just the way I like it.

I still don't know why my previous installation refused to allow EFDI to be installed.

Thnaks again.

Barabazs, 2022-04-24:
Glad to hear it. Too bad you had to resort to that.

closed 2022-04-24 by Barabazs