AdnanHodzic/displaylink-debian #706: "Extra" Intel outputs detected but not DisplayLink

AngelBlue01, 2022-04-10:
My DisplayLink monitor is not being detected, in fact it does not appear that the DisplayLink provider is being detected at all. However xrandr detects 5 providers, which is odd as my integrated Intel graphics card only has 3 outputs, DVI, HDMI (in use) and VGA.

ARandR seems to detect what it labels as following:

- VGA-1
- HDMI-1
- DP-1
- HDMI-2 (in use)
- DP-2
- DVI-I-4-4
- DVI-I-3-3

Except for the one output that's in use, HDMI-2, the others are greyed out and cannot be accessed. Where are these "extra" providers coming from?

I already tried the tips [suggested](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348) in the post-installation guide, namely modifying /etc/X11/xorg.conf.d/20-displaylink.conf and then deleting /etc/X11/xorg.conf.d/20-displaylink.conf, but they made no difference. I have kept the modified /etc/X11/xorg.conf.d/20-displaylink.conf

The most frustrating thing is that the displaylink-debian driver used to work on the same system with the same hardware configuration when I ran Debian 10.x. After I upgraded to 11.1 a few months ago, the driver stopped working, as I expected, but then this issue arose when I tried to reinstall the driver using the latest version. Unfortunately I don't remember what configuration in /etc/X11/xorg.conf.d/20-displaylink.conf was working with Debian 10.x.

Thanks.

Here is the contents of the `debug` command:

> Starting Debug ...
> 
> Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
> 
> Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
> 
> --------------- Linux system info ----------------
> 
> Distro: Debian
> Release: bullseye
> Kernel: 5.10.0-13-amd64
> 
> ---------------- DisplayLink info ----------------
> 
> Driver version: 1.10.1
> DisplayLink service status: up and running
> EVDI service version: 1.10.1
> 
> ------------------ Graphics card -----------------
> 
> Vendor: i915
> Subsystem: 2nd
> VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
> VGA (3D): 
> X11 version: 1.20.11-1+deb11u1
> X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xrdp/xorg.conf
> 
> -------------- DisplayLink xorg.conf -------------
> 
> File: /etc/X11/xorg.conf.d/20-displaylink.conf
> Contents:
>  Section "OutputClass"
>     Identifier  "DisplayLink"
>     MatchDriver "evdi"
>     Driver      "modesetting"
>     Option      "AccelMethod" "none"
> EndSection
> 
> -------------------- Monitors --------------------
> 
> Providers: number : 5
> Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 5 associated providers: 1 name:modesetting
> Provider 1: id: 0xe8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
> Provider 2: id: 0xc6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
> Provider 3: id: 0xa4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
> Provider 4: id: 0x82 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

Barabazs, 2022-04-19:
You have to associate all the providers to provider 0. You only associated provider 1 in your output.

AngelBlue01, 2022-04-23:
That worked! Thank you!

Is this something I will need to do on every startup or is simple done once?

Barabazs, 2022-04-24:
iirc it has to be done once. Otherwise you can make an alias or start-up rule
https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#automated-persistent-display-configuration

closed 2022-04-24 by Barabazs